ReFirmLabs/binwalk #359: Please add pip install instructions to README page

E3V3A, 2018-10-13:
Hi! 
Can someone please add the *pip install** instructions to the readme page?

This may be related to #352.

E3V3A, 2018-10-14:
For example, is `python-lzma` the same as `pip3 install pylzma` for P3?

E3V3A, 2018-10-15:
I was able to get most of it going with:
```bash
pip3 install cycler kiwisolver matplotlib
pip3 install pylzma
pip3 install nose coverage
pip3 install pycryptodome
pip3 install pyqtgraph
pip3 install capstone
pip3 install cstruct
pip3 install ubi_reader
```
But that method is missing all the qt4 and opengl stuff, as normally installed with:
```bash
# (1) sudo apt-get install libqt4-opengl python3-opengl python3-pyqt4 python3-pyqt4.qtopengl
# (2) sudo apt-get install python3-numpy python3-scipy python3-pip
```
I have no idea what would be the equivalent pip3 packages for (1), if any.
Therefore you will not be able to get any graphical output.

chankruze, 2019-03-17:
You are using a MAC ?

0xd3xt3r, 2019-09-01:
*pip3 install ubi_reader* this will not work. 
*ubi-reader* library is on python 2 not ported on python3 yet.

closed 2024-09-23 by devttys0