ReFirmLabs/binwalk #363: Add modified CramFS to filesystem magic

E3V3A, 2018-10-16:
Some time ago, there were a bunch of *Pirelli* routers whose firmware was using a **modified CramFS**.
(also containing LZMA structures).

The file magic for these were:  **`45 3D CD 28`**.
With LZMA:
 - major blocks using: `5D 00 00 40` and 
 - minor ones using `5D 00 00 80`

I suggest to call this after the modifers:
`Jungo's OpenRG version of CramFS` or alternatively 
`Jungo modified CramFS`

How can I add this to the magic file?

E3V3A, 2018-11-26:
uuhm, this is not a `support` ticket, it's an `enhancement`.

devttys0, 2018-11-27:
Since the LZMA version of CramFS has the same signature, I think there are two approaches to this.

First, the easy way: just add the lzma-uncramfs utility as a cramfs extractor. If it fails, binwalk will move on to the next cramfs extractor. This requires minimal effort, and the files will be extracted correctly, but there won't be anything in binwalk's output to indicate that the CramFS image has Jungo's LZMA modifications.

Second, the "correct" way: write a plugin. I don't think this can be done with a simple signature as the location of the compressed blocks is not necessarily predictable (at least AFAIK). A plugin that does some basic parsing of the CramFS image and looks at a compressed block and sees that it starts with `0x5D` could then modify the displayed description string to indicate that it's a Jungo LZMA CramFS image.