ReFirmLabs/binwalk #365: Funny QNX6 file system magic

E3V3A, 2018-10-17:
I've been looking at an issue why a Samsung FW blob has a QNX6 FS in it.

**I am sure this is not true**, since the magic doesn't align while the blob is huge and likely to be a false positive. So I looked into the QNX6 magic as defined in the latest [filesystems](https://github.com/ReFirmLabs/binwalk/blob/798ac5a48e2463fb44d254e0db62befd0d538507/src/binwalk/magic/filesystems#L684-L685) file:

```magic

# QNX6 filesystem
0 string \xEB\x10\x90\x00 QNX6 filesystem
```

However, after reading this [Forensic Focus article](https://www.forensicfocus.com/Forums/viewtopic/t=16846/) and looking at the corresponding [QNX6 extractor](https://nop.ninja/qnx6-0.2d.tar.xz), it seem that there is a different _magic_ available, namely `\68\19\11\22`.

```python
class QNX6FS:

	PARTITION_MAGIC	= {'QNX4':0x002f,'QNX6':0x68191122}
	FILE_TYPE 		= {'DIRECTORY':0x01,'DELETED':0x02,'FILE':0x03}

	QNX6_SUPERBLOCK_SIZE 	= 0x200 	#Superblock is fixed (512 bytes)
	QNX6_SUPERBLOCK_AREA 	= 0x1000	#Area reserved for superblock
	QNX6_BOOTBLOCK_SIZE  	= 0x2000	#Boot Block Size
	QNX6_DIR_ENTRY_SIZE  	= 0x20 		#Dir block size (32 bytes)
	QNX6_INODE_SIZE      	= 0x80 		#INode block size (128 bytes)
	QNX6_INODE_SIZE_BITS 	= 0x07 		#INode entry size shift

	QNX6_NO_DIRECT_POINTERS = 16		#Max Direct iNodes
	QNX6_PTR_MAX_LEVELS 	= 5			#Max Indirect iNodes
	QNX6_SHORT_NAME_MAX 	= 27		#Short Name Max Length
	QNX6_LONG_NAME_MAX 		= 510		#Long Name Max Length

	def __init__(self, source):
		self.TARGET_ = source
		
	def GetPartitions(self):
		with open(self.TARGET_, "rb") as handle:
			DataBlock = handle.read(512);

			##Split DataBlock into parts
			BootCode = DataBlock[0:446]
			MasterPartitionTable = DataBlock[446:510]
			BootRecordSignature = DataBlock[510:512]

			##Detect if MBR is valid.
			BootRecordSignature = unpack('H', BootRecordSignature)[0]
			if BootRecordSignature == 0xAA55:
				print "[-] BootRecordSignature Detected."
			else:
				raise IOError('[ERROR] BootRecordSignature Missing; Invalid Disk Image')
				exit()
			return self.parsePartitionMBR(handle,0)
...
```

I think we need to change this and include it. 
Also, there are no references in the magic file, as to where this info was obtained. 

Any ideas how to proceed?

E3V3A, 2018-10-17:
Here's the `QNX6_SUPER_MAGIC` definition in linux:
 
https://github.com/torvalds/linux/blob/6f0d349d922ba44e4348a17a78ea51b7135965b1/include/uapi/linux/magic.h#L49
https://github.com/torvalds/linux/blob/master/fs/qnx6/inode.c

KuroSaru, 2018-10-17:
E:V:A.

This issue is that current magic has a definition for the boot block of QNX4 but not the magic number for QNX6.

E3V3A, 2018-10-17:
@KuroSaru 
Hi! Please let me understand what you are saying.
 1. So are you saying that `\xEB\x10\x90\x00`, is for a QNX4 *boot* block?
 2. That we need to add: `\x68\x19\x11\x22` for a "super block". 
 3. What is a "super block"? A partition?
 4. What are the signatures for the files contained within, then?

KuroSaru, 2018-10-18:
Hi,

1. Yes that is a QNX4 bootblock.
2/3. The added value is for the QNX6 superblock, this is used to find all
filess in a QNX file system, a superblock in terms of QNX is same as
EXT2/3/4 it is required to parse and process the filesystems.
4: There are no signautres that determine a single file in QNX as they are
split up acorss multiple blocks, it requires a iNode structure to process
and get a block.

Hope This Helps.

E3V3A, 2018-10-19:
@KuroSaru 
Thanks for that info. Do you have a small example file of a QNX6 blob bin, we can use to test this with? 

@devttys0 
What is the best way to use a plugin to parse these files?

E3V3A, 2018-10-19:
I just added the updated magic in PR #366 but we would probably need @KuroSaru  help to get a proper extended magic that also show block sizes and locations.

E3V3A, 2018-11-26:
If you merge #366 and then #371 we can close this!

CoffeeExpress, 2018-11-27:
Done and done!

closed 2018-11-27 by CoffeeExpress

Gerd1986, 2020-07-28:
> > I've been looking at an issue why a Samsung FW blob has a QNX6 FS in it.
> > **I am sure this is not true**, since the magic doesn't align while the blob is huge and likely to be a false positive. So I looked into the QNX6 magic as defined in the latest [filesystems](https://github.com/ReFirmLabs/binwalk/blob/798ac5a48e2463fb44d254e0db62befd0d538507/src/binwalk/magic/filesystems#L684-L685) file:
> > ```
> > 
> > # QNX6 filesystem
> > 0 string \xEB\x10\x90\x00 QNX6 filesystem
> > ```
> > 
> > 
> > However, after reading this [Forensic Focus article](https://www.forensicfocus.com/Forums/viewtopic/t=16846/) and looking at the corresponding [QNX6 extractor](https://nop.ninja/qnx6-0.2d.tar.xz), it seem that there is a different _magic_ available, namely `\68\19\11\22`.
> > ```python
> > class QNX6FS:
> > 
> > 	PARTITION_MAGIC	= {'QNX4':0x002f,'QNX6':0x68191122}
> > 	FILE_TYPE 		= {'DIRECTORY':0x01,'DELETED':0x02,'FILE':0x03}
> > 
> > 	QNX6_SUPERBLOCK_SIZE 	= 0x200 	#Superblock is fixed (512 bytes)
> > 	QNX6_SUPERBLOCK_AREA 	= 0x1000	#Area reserved for superblock
> > 	QNX6_BOOTBLOCK_SIZE  	= 0x2000	#Boot Block Size
> > 	QNX6_DIR_ENTRY_SIZE  	= 0x20 		#Dir block size (32 bytes)
> > 	QNX6_INODE_SIZE      	= 0x80 		#INode block size (128 bytes)
> > 	QNX6_INODE_SIZE_BITS 	= 0x07 		#INode entry size shift
> > 
> > 	QNX6_NO_DIRECT_POINTERS = 16		#Max Direct iNodes
> > 	QNX6_PTR_MAX_LEVELS 	= 5			#Max Indirect iNodes
> > 	QNX6_SHORT_NAME_MAX 	= 27		#Short Name Max Length
> > 	QNX6_LONG_NAME_MAX 		= 510		#Long Name Max Length
> > 
> > 	def __init__(self, source):
> > 		self.TARGET_ = source
> > 		
> > 	def GetPartitions(self):
> > 		with open(self.TARGET_, "rb") as handle:
> > 			DataBlock = handle.read(512);
> > 
> > 			##Split DataBlock into parts
> > 			BootCode = DataBlock[0:446]
> > 			MasterPartitionTable = DataBlock[446:510]
> > 			BootRecordSignature = DataBlock[510:512]
> > 
> > 			##Detect if MBR is valid.
> > 			BootRecordSignature = unpack('H', BootRecordSignature)[0]
> > 			if BootRecordSignature == 0xAA55:
> > 				print "[-] BootRecordSignature Detected."
> > 			else:
> > 				raise IOError('[ERROR] BootRecordSignature Missing; Invalid Disk Image')
> > 				exit()
> > 			return self.parsePartitionMBR(handle,0)
> > ...
> > ```
> > 
> > 
> > I think we need to change this and include it.
> > Also, there are no references in the magic file, as to where this info was obtained.
> > Any ideas how to proceed?
> 
> @E3V3A @KuroSaru sorry for commenting on a closed issue but I could not find contact info for either of you. The nop.ninja site is down, any chance either of you have a copy of the `qnx6-0.2d.tar.xz` file? I would like to investigate integrating this into binwalk.

Hey eacmen,

[qnx6-0.2d.zip](https://github.com/ReFirmLabs/binwalk/files/4988991/qnx6-0.2d.zip)

et voila.

Greetings Gerd

eacmen, 2020-07-28:
> 
> Hey eacmen,
> 
> [qnx6-0.2d.zip](https://github.com/ReFirmLabs/binwalk/files/4988991/qnx6-0.2d.zip)
> 
> et voila.
> 
> Greetings Gerd

Gerd,

merci!

-peter