ReFirmLabs/binwalk #368: Unable to extract ISO 9660 file

PerezFederico, 2018-11-08:
[target.iso.zip](https://github.com/ReFirmLabs/binwalk/files/2563109/target.iso.zip)

This file seems to spin or lock binwalk forever.
It already happened to me with several ISO files along the way (I use binwalk a lot).
It may be a El Torito extension to ISO 9660, but I'm not sure.
I'm would be happy to contribute to binwalk if this is a confirmed bug and point in the right direction.

Thanks
Fede

mzpqnxow, 2018-11-25:
@0xfede7c8 is this spinning in Python or a third party native code app? Can you control-c to get a backtrace if this is spinning in Python?

devttys0, 2018-11-27:
@0xfede7c8 Please provide the exact command line that you're running, as binwalk runs through this ISO image in about 6 seconds on my box.

0xfede7c8, 2018-11-27:
Forgot to mention. I was trying to extract it.

    binwalk --extract su8sv120.iso

zazazo, 2018-12-14:
Same issue here.

0xfede7c8, 2018-12-18:
Found another file that hangs binwalk when tried to extract it:

[LAT_A06.zip](https://github.com/ReFirmLabs/binwalk/files/2691265/LAT_A06.zip)

This one is with LHa compression

devttys0, 2019-01-22:
I pushed some commits which addressed issues with both ISO and LHA extraction. Should work now. :)