ReFirmLabs/binwalk #370: Error while running binwalk

mzfr, 2018-11-11:
I am using binwalk on Manjaro. I have tried to install it multiple times but every time I am getting this error:
```
➜ binwalk
Traceback (most recent call last):
  File "/bin/binwalk", line 25, in <module>
    import binwalk
  File "/lib/python3.7/site-packages/binwalk/__init__.py", line 3, in <module>
    from binwalk.core.module import Modules
  File "/lib/python3.7/site-packages/binwalk/core/module.py", line 10, in <module>
    import inspect
  File "/usr/lib/python3.7/inspect.py", line 40, in <module>
    import linecache
  File "/usr/lib/python3.7/linecache.py", line 11, in <module>
    import tokenize
  File "/usr/lib/python3.7/tokenize.py", line 33, in <module>
    import re
  File "/usr/lib/python3.7/re.py", line 143, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
```

E3V3A, 2018-11-26:
@mzfr 
You need to provide more info, before anyone can help you.
* How do you do about installing? 
* What packages have you installed?
* Are they up-to-date?

Please post your output from doing: 
1. Do `pip install pip-date` and run `pip-date` and post your output.
2. Do `pip list --outdated` and post output.

sundhaug92, 2018-11-26:
Python 3.7? I'm not sure that's supported

devttys0, 2018-11-27:
@sundhaug92 https://stackoverflow.com/questions/43124775/why-python-3-6-1-throws-attributeerror-module-enum-has-no-attribute-intflag

devttys0, 2018-11-27:
Whoops, meant to tag @mzfr in that last comment. Anyway, the issue doesn't appear to be with binwalk; in fact no real code has actually executed at the time of the exception, it's just failing to import some standard modules that binwalk relies on. Definitely sounds like an issue with conflicting python module names, as referenced in the SO link above.

E3V3A, 2018-11-27:
so why not close the issue?

mzfr, 2018-11-28:
Yes I'll close the issue, I got it working with python 2.

closed 2018-11-28 by mzfr