ReFirmLabs/binwalk #380: Cannot indentify uImage header in Asus FW

Crankle437, 2019-01-14:
Hi,
binwalk (v2.1.2-fd55d7f) cannot identify uImage Header in Asus firmwares.
I've tried it with this Asus firmware: https://www.asus.com/Networking/RTAC55UHP/HelpDesk_BIOS/
Header is a typically uImage Header and 64 Bytes long: 

> 27 05 19 56

devttys0, 2019-01-23:
Binwalk sees the uImage header, but flags it as invalid because it seems ASUS has used the `name` field of the uImage header to store some binary data (the `name` field starts with `03 00 00 04`, followed by the actual ASCII name). I guess there technically isn't any restriction on mixing binary data with ASCII data in that field, but that's not the intended use of the `name` field, which is why binwalk thinks that uImage header is a false positive. I'll have to see what the easiest way to add support for this is.

devttys0, 2024-09-23:
Binwalk v3 seems to handle these correctly:

[image]

closed 2024-09-23 by devttys0