ReFirmLabs/binwalk #388: Doesn't work on Windows

bannsec, 2019-02-18:
In the latest release notes you mention that it should work on Windows. However, after pip installing it i get the following errors:

```
In [2]: binwalk.scan(r"C:\Windows\System32\Calc.exe", signatures=True)

General Error: Cannot open file : [Errno 2] No such file or directory: '--signatures'

---------------------------------------------------------------------------
ModuleException                           Traceback (most recent call last)
<ipython-input-2-a05707c228ff> in <module>
----> 1 binwalk.scan(r"C:\Windows\System32\Calc.exe", signatures=True)

c:\users\loki\miniconda3\envs\stegoveritas\lib\site-packages\binwalk\__init__.py in scan(*args, **kwargs)
      8 def scan(*args, **kwargs):
      9     with Modules(*args, **kwargs) as m:
---> 10         objs = m.execute()
     11     return objs
     12

c:\users\loki\miniconda3\envs\stegoveritas\lib\site-packages\binwalk\core\module.py in execute(self, *args, **kwargs)
    782         # Run all modules
    783         for module in self.list():
--> 784             obj = self.run(module)
    785
    786         # Add all loaded modules that marked themselves as enabled to the

c:\users\loki\miniconda3\envs\stegoveritas\lib\site-packages\binwalk\core\module.py in run(self, module, dependency, kwargs)
    801         '''
    802         try:
--> 803             obj = self.load(module, kwargs)
    804
    805             if isinstance(obj, binwalk.core.module.Module) and obj.enabled:

c:\users\loki\miniconda3\envs\stegoveritas\lib\site-packages\binwalk\core\module.py in load(self, module, kwargs)
    832         argv = self.argv(module, argv=self.arguments)
    833         argv.update(kwargs)
--> 834         argv.update(self.dependencies(module, argv['enabled']))
    835         return module(self, **argv)
    836

c:\users\loki\miniconda3\envs\stegoveritas\lib\site-packages\binwalk\core\module.py in dependencies(self, module, module_enabled)
    864             # raise an exception
    865             if depobj.errors:
--> 866                 raise ModuleException("Failed to load " + dependency.name + " module")
    867             else:
    868                 attributes[dependency.attribute] = depobj

ModuleException: Failed to load General module
```

devttys0, 2019-02-19:
Windows support for binwalk is experimental at best (only tested on Windows XP with Python 2.7, and even that was a long time ago), and the API has not been tested on Windows at all. Do you get the same error when running binwalk from the command line?

bannsec, 2019-02-20:
So after pip installing it, it doesn't appear to even be in my path on Windows.

einstein95, 2019-04-08:
Binwalk installation from pip is not supported, see https://github.com/ReFirmLabs/binwalk/issues/121#issuecomment-244626235

Bamgm14, 2019-09-14:
To Be Prefectly Honest. It works fine on windows 10 python 3.7.2 64 bits and 3.7 32 bits. Its just that the binwalk file isn't recognized as python for some reason. I copied that file and used it a the directories i needed it. Worked like a charm
Thanks for the program.

akx, 2020-04-12:
#454 may fix some of this stuff. With it, I'm at least able to successfully run the tests on Windows (Python 3).

rajhlinux, 2020-04-28:
@Bamgm14 Can you write a tutorial on how you got it working on windows 10?
Would be great.
Thanks.

Bamgm14, 2020-04-28:
Well. First, Download the file from this Repository and run the setup command. You can also do pip but then you will have manuelly add some missing files (IDK why the core and few other stuff are missing @devttys0, please update the pip installer for Windows <3). Then once you have done that, if you used pip, copy all the files from src/binwalk into the %Whatever directory python.exe is in%\Lib\site-packages\binwalk. Once that is done, got to the %Whatever directory python.exe is in%\Scripts and copy the binwalk file. It should not have an extension on it. Or you can just copy it from src\scripts folder from here. Then just run python binwalk and you should get list of commands. It should work then. I am trying to find away to run it like in Linux but to no success.
Anyway, hope this helps.

Kungergely, 2020-08-09:
Actually the PR by @akx makes this work out of the box on Windows 10. You just have to clone his branch and run `pip install .` in the newly-created folder. Then binwalk can be run either by typing `binwalk` in the command like (if the Python installation's Scripts folder is added to %PATH%) or by typing `python3 -m binwalk`

closed 2024-09-23 by devttys0