ReFirmLabs/binwalk #389: From bytes not str

bannsec, 2019-02-18:
For python3, everything is by default bytes and best practice is to use bytes for all internal representations and handling. I'd like to use the binwalk `string=True` option, but it does not support bytes. This becomes more important given the ambiguity of the files I'd like to binwalk being things that I do not know ahead of time their proper encoding, and so converting from bytes to str is, at best, a guessing game.

The solution for the time being is to write the file to disk using 'wb', then call binwalk as normal, but would be nice to cut out that write to disk step.

bannsec, 2019-02-18:
Also would be reasonable to allow a file-like object as a param, similar to how zip and company do it in core python libraries. That approach means that you wouldn't have to change any backend handling (aside from initially allowing the file like object of course), and the user could decide to open a file themselves, or use BytesIO/StringIO/etc.

devttys0, 2019-02-19:
The `binwalk.scan` method expects string arguments as they would appear from `sys.argv`, which is a `str` type in both Python2 and Python3.

Converting from a `bytes` to a `str` type can be done by decoding the `bytes` object as `latin-1`; knowing the actual encoding type is not necessary as `latin-1` encoding is a direct byte-for-byte mapping:

```
import binwalk
data = b"some-test-data-to-scan"
binwalk.scan(data.decode('latin-1'), string=True)
```

The only downside to this is that if `data` is a `str` and not `bytes`, then it will have no `decode` attribute. For maximum flexibility, you can use the `binwalk.core.common.bytes2str` function; you can pass this function either `str` or `bytes` and it will return the data as a `str` type:

```
import binwalk
from binwalk.core.common import bytes2str
data = b"some-test-data-to-scan"
binwalk.scan(bytes2str(data), string=True)
```

russdill, 2024-01-16:
`Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/binwalk/core/module.py", line 595, in main
    retval = self.run()
             ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/binwalk/modules/signature.py", line 192, in run
    self.scan_file(fp)
  File "/usr/lib/python3/dist-packages/binwalk/modules/signature.py", line 141, in scan_file
    (data, dlen) = fp.read_block()
                   ^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/binwalk/core/common.py", line 517, in read_block
    data = self.read(self.block_read_size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/binwalk/core/common.py", line 483, in read
    data += tmp
TypeError: can't concat str to bytes`

I've tried this, it doesn't work. The read function starts with a b'' and attempts to concatenate data from StringFile which is str.