ReFirmLabs/binwalk #392: Cannot indentify CSR Bluecore firmware segment

ChoKyuWon, 2019-03-04:
HI!
My binwalk(V2.1.1) cannot identify CSR Bluecore firmware segment.
The magic word is CSRbcfw1, and It's in my file, but binwalk dosen't work.

Here is my file : https://drive.google.com/open?id=1xWF6pYdV-CAuFMDE-PmjHCkXhplkVCgy

And Here is my binwalk result:
```
$ binwalk xpv_little.bin
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
```

chankruze, 2019-03-14:
No result at all ?
Let me try

ChoKyuWon, 2019-03-14:
@chankruze  Yes, No result. 
[xpv_little.zip](https://github.com/ReFirmLabs/binwalk/files/2966723/xpv_little.zip)
I double check that my file contain `CSRbcfw1` and get no result on binwalk.

chankruze, 2019-03-15:
@ChoKyuWon 
yes, i found `CSRbcfw1` in Linear Disassembly between location `00000660` and `00000680`.Open attached file in hex editer.
[xdv_little.txt](https://github.com/ReFirmLabs/binwalk/files/2972022/xdv_little.txt)
 So we need a response here.

ChoKyuWon, 2019-03-15:
@chankruze  Well, I think I upload wrong file on first comment. firmware is on XPV, not XDV.

I think XDV is just data file, not firmware segment. Maybe `u2CSRbcfw1CSR-dfu1CSR-dfu2L` is just showed as firmware version.

In XPV,  magic word `CSRbcfw1` is on `10000`.
[image]
I want to know that binwalk is also not work on XPV file.

In summarize,

1.  `binwalk` is not work on XDV file(`xdv_little.bin`) and file contain `u2CSRbcfw1CSR-dfu1CSR-dfu2L`  on `00000660` to `00000680`
2.  Is `binwalk` also not work on XPV file(`xpv_little.zip`)? It contain `CSRbcfw1` on `00010000` and in my environment, `binwalk` is not work.

chankruze, 2019-03-15:
Yes i checked XPV file, `CSRbcfw1` on `00010000`.

I have a a Minix image, firmware mod kit is unable to extract it, any suggestion ?
the repo for that image is https://github.com/chankruze/Intex_Eco_Beats

ChoKyuWon, 2019-03-15:
@chankruze 
I can't understand what did you talking about.(because my poor english, maybe).
Do you mean your Minix image contain bluetooth firmware but firmware mod kit can't extract it?

If you mean this, I have no idea. Bluetooth use processer named `XAP Familly`, and it's disassembler is so hard to find.
Maybe this IDA plugin might help : https://github.com/comex/xap, but I'm not sure.

If you mean another thing, please tell me more specifically.

Thanks,