ReFirmLabs/binwalk #394: Cannot identify firmware FS nor kernel

ub3rduck, 2019-03-17:
I'm trying to unpack DLink DSL-2730U H/W:V1 firmware using binwalk.
Dlink provides multiple files when you download the firmware, but the interesting ones are router.img and webupload.img, while the webupload.img is the same as router.img only it contains a digital signature.

When running binwalk on both files it only identifies some 31 LZMA compressed data segments and few GIF images. the decompressed files contains 7z files which are in turn compressed in infinite loop. 
My guess is some kind of encryption? but I don't know where to start since the bootloader is also compressed and I don't have the original decompressed fw to start with. 

`firmware/dlink/DSL-2730U$ binwalk webupload.img 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
73            0x49            LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32729 bytes
8172          0x1FEC          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32752 bytes
13856         0x3620          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32741 bytes
19210         0x4B0A          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32757 bytes
26282         0x66AA          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32750 bytes
32344         0x7E58          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32687 bytes
40936         0x9FE8          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32744 bytes
49822         0xC29E          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32697 bytes
57147         0xDF3B          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32697 bytes
62616         0xF498          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32748 bytes
68427         0x10B4B         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32706 bytes
75350         0x12656         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32735 bytes
81129         0x13CE9         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32757 bytes
86800         0x15310         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32725 bytes
93039         0x16B6F         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32751 bytes
98928         0x18270         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32720 bytes
104323        0x19783         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32751 bytes
109419        0x1AB6B         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32735 bytes
115198        0x1C1FE         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32744 bytes
121541        0x1DAC5         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32700 bytes
128428        0x1F5AC         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32757 bytes
134657        0x20E01         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32744 bytes
141025        0x226E1         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32758 bytes
146871        0x23DB7         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32741 bytes
152830        0x254FE         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32704 bytes
157806        0x2686E         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32748 bytes
163322        0x27DFA         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32708 bytes
169117        0x2949D         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32724 bytes
175171        0x2AC43         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32748 bytes
181584        0x2C550         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32758 bytes
188203        0x2DF2B         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 11829 bytes
191236        0x2EB04         GIF image data, version "89a", 12 x 12
191330        0x2EB62         GIF image data, version "89a", 12 x 12
191410        0x2EBB2         GIF image data, version "89a", 836 x 92
193762        0x2F4E2         GIF image data, version "89a", 114 x 35
195420        0x2FB5C         GIF image data, version "89a", 125 x 25
`

Any ideas?

chankruze, 2019-03-17:
@ub3rduck upload your image in a zip file here, i will check if i can unpack or not

ub3rduck, 2019-03-18:
@chankruze 
https://dlinkmea.com/upload/downloadable/Firmware_57e21e8a9469e.rar

thank you. waiting for your insights

chankruze, 2019-03-18:
The filesystem is not supported
```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
73            0x49            LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32729 bytes
8172          0x1FEC          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32752 bytes
13856         0x3620          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32741 bytes
19210         0x4B0A          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32757 bytes
26282         0x66AA          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32750 bytes
32344         0x7E58          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32687 bytes
40936         0x9FE8          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32744 bytes
49822         0xC29E          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32697 bytes
57147         0xDF3B          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32697 bytes
62616         0xF498          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32748 bytes
68427         0x10B4B         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32706 bytes
75350         0x12656         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32735 bytes
81129         0x13CE9         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32757 bytes
86800         0x15310         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32725 bytes
93039         0x16B6F         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32751 bytes
98928         0x18270         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32720 bytes
104323        0x19783         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32751 bytes
109419        0x1AB6B         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32735 bytes
115198        0x1C1FE         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32744 bytes
121541        0x1DAC5         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32700 bytes
128428        0x1F5AC         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32757 bytes
134657        0x20E01         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32744 bytes
141025        0x226E1         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32758 bytes
146871        0x23DB7         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32741 bytes
152830        0x254FE         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32704 bytes
157806        0x2686E         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32748 bytes
163322        0x27DFA         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32708 bytes
169117        0x2949D         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32724 bytes
175171        0x2AC43         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32748 bytes
181584        0x2C550         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 32758 bytes
188203        0x2DF2B         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 11829 bytes
191236        0x2EB04         GIF image data, version "89a", 12 x 12
191330        0x2EB62         GIF image data, version "89a", 12 x 12
191410        0x2EBB2         GIF image data, version "89a", 836 x 92
193762        0x2F4E2         GIF image data, version "89a", 114 x 35
195420        0x2FB5C         GIF image data, version "89a", 125 x 25

Extracting 0 bytes of  header image at offset 0
ERROR: No supported file system found! Aborting...
```

ub3rduck, 2019-03-19:
Appreciate your effort. thanks for the confirmation

devttys0, 2019-04-03:
@ub3rduck This doesn't appear to be a normal Linux firmware, likely an RTOS of some sort. After Googling some unique strings found in the firmware and some of the extracted LZMA blobs, it's suggested that this is running the ThreadX RTOS:

https://reverseengineering.stackexchange.com/questions/15666/router-flash-dump-unknown-filesystem

https://translate.google.com/translate?hl=en&sl=pt&u=https://ryan.com.br/blogs/quicktalk/2015/09/por-dentro-do-roteador-wi-fi-d-link-dsl2740e/&prev=search

closed 2019-07-17 by CoffeeExpress