ReFirmLabs/binwalk #396: Can't extract YAFFS

0xfede7c8, 2019-03-21:
Sample attached:
[firmware.zip](https://github.com/ReFirmLabs/binwalk/files/2992825/firmware.zip)

When executing:

    binwalk firmware.fl1 --extract -M

it locks forever.

It seems a bug in yaffshiv because it loops in 100% CPU, but it seems logical to post it here as binwalk uses this tool.

0xfede7c8, 2019-03-21:
Just realized that yaffshiv is from one of the main contributors of binwalk. Let me know if you want me to open an issue there.

devttys0, 2019-04-03:
What version of binwalk are you running, and what is binwalk's output without the `--extract` option specified? I don't see anything that looks like a YAFFS file system in that firmware image.

0xfede7c8, 2019-04-03:
Version is:
Binwalk v2.1.2-a0c5315
These are the logs, it gets stuck right there:
[yaffs_binwalk_log.txt](https://github.com/ReFirmLabs/binwalk/files/3038932/yaffs_binwalk_log.txt)

yssreddy1961, 2021-10-10:
the file given above is not a yaffs2 file system
the firmware is  LENOVA UEFI bios firmware
binwalk  will not extract the file
i used uefi-firmware-parser and extracted firmware

platomav, 2022-09-15:
After stumbling across the same (yaffshiv) bug, it seems to be caused because it tries to find the next page by appending the size of the last one without checking if that size is logical. For example, a <ins>false positive</ins> "YAFFS" file like this may have a next page size of 0 which will causes an infinite loop by re-finding the same page.

Anyway, **this appears to have now been fixed** (indirectly) by the recent code changes at https://github.com/devttys0/yaffshiv/pull/2 which were merged at https://github.com/devttys0/yaffshiv/commit/d57d68a225cdc8c4965e96ef05beef58911ff5eb.

For reference, I have attached the actual <ins>false positive</ins> "YAFFS" file which causes the infinite loop at yaffshiv and thus Binwalk.

[14220C.zip](https://github.com/ReFirmLabs/binwalk/files/9575207/14220C.zip)

closed 2024-09-23 by devttys0