ReFirmLabs/binwalk #405: API Issue: Recursive extraction using Binwalk API possible?

HassamK97, 2019-05-09:
Hi, I am using binwalk API to perform firmware extraction. When extraction is performed it only returns some extracted files and not the complete file system. Not like the tool which returns complete file system after recursive extraction. 

`binwalk -M -e dlink_DCS_930L.bin`

This command recursively extracts binary file and displays following information in extracted folder:
[image]

But if I perform extraction from binwalk API, it doesn't extract recursively. The following code is used for performing firmware extraction using API:

`for module in binwalk.scan('dlink_DCS_930L.bin',  signature=True, quiet=True, extract=True):`

It does not display the complete file system of the firmware like above. The result is as follows:
[image]

Is there any rule or any specific method that performs recursive extraction using binwalk Python API? Kindly acknowledge.

Kruxinator, 2019-05-27:
Use
`for module in binwalk.scan('dlink_DCS_930L.bin', signature=True, quiet=True, matryoshka=True, extract=True):`

HassamK97, 2019-05-28:
Thanks, Issue resolved.

HassamK97, 2019-05-28:
> Use
> `for module in binwalk.scan('dlink_DCS_930L.bin', signature=True, quiet=True, matryoshka=True, extract=True):`

Thanks, issue resolved.

closed 2019-05-28 by HassamK97