ReFirmLabs/binwalk #410: binwalk c12fdd7 fails to install - invalid syntax in core/settings.py

voltagex, 2019-06-14:
After a git clone:

```
~/binwalk$ sudo python3 setup.py install
creating /home/voltagex/binwalk/src/binwalk/core/version.py
running install
running build
running build_py
copying src/binwalk/core/version.py -> build/lib/binwalk/core
running build_scripts
running install_lib
copying build/lib/binwalk/core/version.py -> /usr/local/lib/python3.7/dist-packages/binwalk/core
byte-compiling /usr/local/lib/python3.7/dist-packages/binwalk/core/settings.py to settings.cpython-37.pyc
  File "/usr/local/lib/python3.7/dist-packages/binwalk/core/settings.py", line 135
```

Wh1terat, 2019-06-14:
Typo on line 135; "Not" instead of "not"

voltagex, 2019-06-23:
We've now got 3 duplicate pull requests for this, what's going on?

jayluxferro, 2019-06-23:
I guess someone forgot to check the pull request before raising an issue.

devttys0, 2019-07-08:
Yep, that was my fault, I didn't catch (or test) that typo before merging. Fixed in master.

closed 2019-07-08 by devttys0