ReFirmLabs/binwalk #416: Make binwalk be more disk friendly when scanning for raw compression streams

PlasmaHH, 2019-06-27:
When scanning larger firmware images for raw compression streams even big disks quickly get filled by the .deflate files that are left behind. Maybe add an option to remove those files (or make it a default and add an option to keep them, or even add an option to keep then only if the uncompressed result is bigger than X).

Also the further this goes the slower binwalk gets until it needs several seconds per new file, not sure if this is part of this.

devttys0, 2019-07-08:
Use the `--rm` option to remove the `.deflate` files automatically. Not sure about the slow down, as I have not experienced this even with very large (many GB) firmware images. It could be simply be due to the disk filling up, or something worse like a memory leak...

PlasmaHH, 2019-07-11:
I will try again with the --rm option. I did not consider it doing what i want since the docs say "Delete carved files after extraction" which somehow made me think its the extracted files, not the extracted archives.

closed 2020-03-14 by devttys0