ReFirmLabs/binwalk #417: Incomplete squashfs extraction due to its hidden padding feature

secmobi, 2019-07-01:
I got a Squashfs little endian image embedded in a firmware image (in ext2 format). The image's magic number is `hsqs` and the version number in offset 28 is `1`. The size showed in offset 40 is `16197478`. Therefore binwalk extracted 16197478 bytes since the magic number. After that, `unsquashfs` failed to extract the file with error logs like this: 

> Parallel unsquashfs: Using 32 processors
Lseek failed because Invalid argument
read_block: failed to read block @0x7531c71abb53508e
read_uids_guids: failed to read id table block
FATAL ERROR:failed to uid/gid table

I researched on squashfs format for a while. [This write up](https://dr-emann.github.io/squashfs/) mentioned that 

> bytes_used: The number of bytes used by the archive. Because squashfs archives are often padded to 4KiB, this can often be less than the file size

Therefore I rounded up the size `16197478` to its 4K alignment 16199680. Extracted the squashfs image by rounded size 16199680, tried `unsquashfs` again. Bingo! 

Besides, since the squashfs image was included in (not after) a ext2 images, I mounted the ext2 to see what it originally looks like. Its size is indeed 16199680, not 16197478 in its header structure. 

Therefore I'm wondering whether we should trust the size value parsed by signatures from the squashfs image header. 

The firmware could be downloaded from [here](https://download.peplink.com/firmware/br1ac/fw-max_br1mk2_hotspot_sohomk3-7.1.2-build1387.bin) for testing. It's 33MB.

devttys0, 2019-07-08:
I don't think the issue here is the extra padding data, for a few reasons:

1. All bytes after the reported size of 16197478 are null bytes (likely just padding and not used by the SquashFS image at all).

2. Simply running `dd` to extract the full 16199680 bytes of the SquashFS file from the original firmware image results in the same error during extraction.

3. After mounting the EXT file system, I copied off the `rdisk.sfs` SquashFS file, which extracted just fine. I then truncated the `rdisk.sfs` file to the reported size of 16197478, and it still extracted.

The issue seems to be that the because the SquashFS image file is contained inside an EXT image, simply copying the raw data out of the EXT image results in also grabbing EXT metadata, resulting in a corrupt copy of the SquashFS image.

secmobi, 2019-07-09:
@devttys0 You're right. I made a mistake during the testing -- mixed up the `rdisk.sfs` file from EXT image and the manually extracted aligned image. They're in the same size and I renamed them by hash values and that caused the mixing up. Just did the test again and showed the same results as yours. Sorry for the misleading proposal from my own mistake. I think the issue could be closed now.

closed 2019-07-09 by secmobi