ReFirmLabs/binwalk #419: how can i decrypt huawei HG531s v1 config file?

minanagehsalalma, 2019-07-08:
[I have tried to follow this guide](https://hg658c.wordpress.com/2017/12/04/decrypting-configuration-files-from-other-huawei-home-gateway-routers/) but libxmlapi.so didn't have the required ATP_GetInfo1 function so i moved to the next file libhttpapi.so it had
```
ATP_GetInfo1
ATP_GetInfo2 
ATP_GetInfo3 
ATP_GetInfo4
```
so i copied the first two .

then ATP_GetInfo3 from libcfmapi.so file

then ATP_GetInfo4 from libmsgapi.so file .... and when i tried to run the script decode_keystore.py but it gave me

> ValueError: Input strings must be a multiple of 16 in length

.... and i think because the values wasn't in the same length

[The firmware file
](https://www.mediafire.com/file/4x8y97j256uvd04/hg531s1.BIN/file)

CoffeeExpress, 2019-07-10:
Binwalk doesn't support the encryption used by this firmware file.

closed 2019-07-10 by CoffeeExpress

minanagehsalalma, 2019-07-10:
@CoffeeExpress  do you mean it can't extract the files from the bin file or you mean that  it can't decrypt the config file ? 
Cause i was able to extract the files using 7 zip but got an error at the end 
[image]
[image]
and there wasn't a much missing files tho .

CoffeeExpress, 2019-07-11:
It can't decrypt the config file.

minanagehsalalma, 2019-07-11:
@CoffeeExpress  what do suggest cause i really need to decrypt that file and i have asked on a lot of sites and no one did really help !

vido89, 2019-07-12:
Download aescrypt2 and try aescrypt2_huawei.exe

minanagehsalalma, 2019-07-12:
@vido89  i have already tried it and it didn't work 
from the README file it says 
> These models have hw_ctree.xml file:
> 
> hg55x hg612 hg8010 hg8010c92 hg8040 hg8040f hg8045 hg8045v2 hg8110 hg8110c92
> hg8120 hg8120c92 hg8120r hg8120rc92 hg8240 hg8240c93 hg8240f hg8240fcig
> hg8240j hg8240p hg8240r hg8240rc93 hg8240rva hg8240s hg8240t hg8240w hg8242
> hg8242b hg8244t hg8245 hg8245c93 hg8245f hg8245r hg8245t hg8247 hg8247a
> hg8247t hg8447
> 
and in fact the config file that i want to decrypt is .bin

minanagehsalalma, 2019-07-12:
> aescrypt2_huawei.exe 1 defaultcfg.xml 1.xml
> version header of file wrong!
> 
> Press <Enter> to exit.
>

whhw31, 2025-01-29:
Were you able to figure this out, I'm doing this as a project and i hit a dead end, i was able to binwalk and extract squeshfs and edit for root access on uart and ssh but when the router boots there is a checksum error, probably the bootloader checking the squashfs before decompressing it, and the bootloader part is encrypted

minanagehsalalma, 2025-01-29:
@whhw31 

> and the bootloader part is encrypted

you could simply read it when it's loaded into memory ... stuff gets decrypted in that state.

these [may be of help](https://www.pentestpartners.com/security-blog/how-to-do-firmware-analysis-tools-tips-and-tricks/) ... the part about encrypted bin files. 

these too 

https://www.zerodayinitiative.com/blog/2020/2/6/mindshare-dealing-with-encrypted-router-firmware

https://watchfulip.github.io/28-12-24/tp-link_c210_v2.html

whhw31, 2025-01-31:
I don't know a way to read the memory from the device, the  bootloader menu is very limited, or may be you mean by simulation.

xeXistenZx, 2026-10-02:
@minanagehsalalma 
Did you find a way to decrypt it?