ReFirmLabs/binwalk #433: [OSX] binwalk completly failes to extract files on osx

Hardcore-fs, 2019-11-07:
binwalk -e -c /Users/comp_01/jane/bob.bin
`DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
452610        0x6E802         gzip compressed data, maximum compression, has original file name: "browser.js", from Unix, last modified: 2009-12-07 20:23:54
455138        0x6F1E2         HTML document header
464652        0x7170C         GIF image data, version "89a", 80 x 15
509272        0x7C558         GIF image data, version "89a", 1 x 1
584388        0x8EAC4         GIF image data, version "89a", 222 x
748967        0xB6DA7         GIF image data, version "89a", 60 x 20
951710        0xE859E         HTML document header
982914        0xEFF82         GIF image data, version "89a", 30 x 30
1155970       0x11A382        GIF image data, version "89a", 24866 x
1222638       0x12A7EE        GIF image data, version "89a", 22654 x 2739
1425417       0x15C009        GIF image data, version "89a", 55 x 8227
1433720       0x15E078        GIF image data
1536297       0x177129        GIF image data, version "89a", 19581 x 11682
1591196       0x18479C        GIF image data, version "89a", 31523 x 16221
1836865       0x1C0741        HTML document header
1929834       0x1D726A        HTML document header
1987594       0x1E540A        Certificate in DER format (x509 v3), header length: 4, sequence length: 703`

only the .js file is extracted

Jaakkonen, 2021-01-18:
You can extract all files with `--dd='.*'`. The `-e` flag only exports default filetypes. See: https://github.com/ReFirmLabs/binwalk/wiki/Usage#-e---extract

You can maybe add a pull request adding the extraction rules for your file types in [extract.conf](https://github.com/ReFirmLabs/binwalk/blob/master/src/binwalk/config/extract.conf). Or you can also define user-specific extraction config to `~/.config/binwalk/extract.conf`. For Mac OS paths figure out the code from [settings.py](https://github.com/ReFirmLabs/binwalk/blob/master/src/binwalk/core/settings.py). The config files are then used in the [extractor.py](https://github.com/ReFirmLabs/binwalk/blob/master/src/binwalk/modules/extractor.py)

EAGLESAT, 2021-01-18:
I cant extract this kernel
In binwalk

EAGLESAT, 2021-01-18:
extract it corectly my pro and sent me

في الاثنين، 18 جانفي، 2021 19:17 EAGLE SAT <eaglesat2020@gmail.com> كتب:

closed 2024-09-23 by devttys0