ReFirmLabs/binwalk #443: Extraction hangs when extracting an encrypted zip

targodan, 2019-12-18:
The error from #108 seems to be back. When extracting encrypted zip file binwalk hangs without any output. Happened to me when walking through wannacry as contained in theZoo.

You can get around this problem by adding the `-z` parameter but either a password prompt or aborting the extraction automatically would be nice.

pmarrapese, 2020-03-31:
Found a workaround: `unzip` supports passing options via environment variables, so you can specify the password with something like ` UNZIP='-p mypassword' binwalk -Mre encrypted.bin`

devttys0, 2020-10-13:
Should be fixed now: https://github.com/ReFirmLabs/binwalk/blob/6f64ba30b06a021ab60aca35e720bb4f8ce72ba5/src/binwalk/config/extract.conf#L57

closed 2020-10-13 by devttys0