ReFirmLabs/binwalk #450: Zero output from ergodox hex

johnrichardrinehart, 2020-02-06:
I was trying to analyze some firmware destined for my Ergodox EZ keyboard (downloadable [here](https://oryx.nyc3.digitaloceanspaces.com/NXzB5pejgfegKFsG3p1mYCxu?response-content-disposition=inline%3B%20filename%3D%22ergodox_ez_john-rinehart-s-custom-layout-fo_XqJOo_jOBga.hex%22%3B%20filename%2A%3DUTF-8%27%27ergodox_ez_john-rinehart-s-custom-layout-fo_XqJOo_jOBga.hex&response-content-type=application%2Foctet-stream&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=KVD37DSS7HWQBFAXQONB%2F20200206%2Fnyc3%2Fs3%2Faws4_request&X-Amz-Date=20200206T202712Z&X-Amz-Expires=604800&X-Amz-SignedHeaders=host&X-Amz-Signature=b62c7b09d2c598866106418223a9f6d33b8b3ae3a9c4588e54b194d44a8a90e8) for 7 days). However, I only obtain non-empty output when I use the `-I` flag. I understand that the [magic file](https://github.com/ReFirmLabs/binwalk/wiki/Frequently-Asked-Questions#binwalk-doesnt-recognize-xyz-file-type) may not be detected. However, `file <firware_file_name>.hex` returns 
```
$ file edfw.hex
edfw.hex: ASCII text, with CRLF line terminators
```
And, I think that ASCII text is analyzable by `binwalk`.

I guess I'm wondering if `binwalk` is useful in my case and, if so, what I need to do in order to apply it to my study of the firmware that I flash to my keyboard.

Addendum: I've uploaded the ASCII file to [pastebin](https://pastebin.com/h2KCpvB9) as an alternative source.

devttys0, 2020-03-12:
The contents of this firmware are represented in an ASCII hexadecimal format, specifically, the Intel hex ("ihex") format, hence why the `file` utility tells you that it's an ASCII file. The data represented by those ASCII codes are executable instructions and program data, which must be converted back to a binary format before the processor can execute the firmware. 

Since binwalk was designed for analyzing binary files, you need to convert the firmware to its binary format before analyzing it. This can be done with the Linux `objcopy` tool:

```objcopy --input-target=ihex --output-target=binary edfw.hex edfw.bin```

Now at this point, you have the firmware, all the code and data is there, so a disassembler would be much more useful to you than binwalk. However, if you do an opcode scan with binwalk it does at least give you an idea as to what type of processor the code was compiled for:

```
$ binwalk edfw.bin -A

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
2357          0x935           AVR8 instructions, function prologue
4573          0x11DD          AVR8 instructions, function prologue
6123          0x17EB          AVR8 instructions, function prologue
6619          0x19DB          AVR8 instructions, function prologue
7075          0x1BA3          AVR8 instructions, function prologue
8365          0x20AD          AVR8 instructions, function prologue
8477          0x211D          AVR8 instructions, function prologue
8569          0x2179          AVR8 instructions, function prologue
8781          0x224D          AVR8 instructions, function prologue
8945          0x22F1          AVR8 instructions, function prologue
9081          0x2379          AVR8 instructions, function prologue
9357          0x248D          AVR8 instructions, function prologue
10385         0x2891          AVR8 instructions, function prologue
11843         0x2E43          AVR8 instructions, function prologue
12713         0x31A9          AVR8 instructions, function prologue
13657         0x3559          AVR8 instructions, function prologue
17207         0x4337          AVR8 instructions, function prologue
17419         0x440B          AVR8 instructions, function prologue
18289         0x4771          AVR8 instructions, function prologue
18693         0x4905          AVR8 instructions, function prologue
21365         0x5375          AVR8 instructions, function prologue
21545         0x5429          AVR8 instructions, function prologue
21639         0x5487          AVR8 instructions, function prologue
22007         0x55F7          AVR8 instructions, function prologue
```

closed 2020-03-12 by devttys0

johnrichardrinehart, 2020-03-12:
That was really helpful, particularly the point addressing this file's format as an [Intel HEX file](https://en.wikipedia.org/wiki/Intel_HEX). Thanks for taking the time to explain this.