ReFirmLabs/binwalk #457: binwalk gets stuck while extracting files

Reverier-Xu, 2020-04-25:
when i tried to extract a rar file from a picture, binwalk no longer outputs after scanning out image part just like this:

```
$ binwalk -e test.png

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             JPEG image data, JFIF standard 1.01
30            0x1E            TIFF image data, big-endian, offset of first image directory: 8
```

It will not continue to work unless I hit the enter key in the terminal:

```
$ binwalk -e test.png

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             JPEG image data, JFIF standard 1.01
30            0x1E            TIFF image data, big-endian, offset of first image directory: 8
[Enter key pressed]
137448        0x218E8         RAR archive data, version 4.x, first volume type: MAIN_HEAD
```

But when I used it without '-e', it works well:

```
$ binwalk test.png

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             JPEG image data, JFIF standard 1.01
30            0x1E            TIFF image data, big-endian, offset of first image directory: 8
137448        0x218E8         RAR archive data, version 4.x, first volume type: MAIN_HEAD

```

I'm trying to use binwalk in my python project, but this bug puzzled me. Since my project is a software with a graphical interface, it is impossible for me to send a carriage return to the terminal while it is running. so it just halted.

My platform:

```
 OS: Arch Linux x86_64 
 Host: AERO 15 Classic-XA 
 Kernel: 5.6.6-arch1-1 
 Packages: 2015 (pacman) 
 Shell: fish 3.1.0 
 Resolution: 1920x1080 
 DE: Plasma 
 WM: KWin 
```

KanoWill, 2020-07-30:
I meet the problem too ,and I'm eager to get the solution.
My environment: Kali Linux latest version arm64.

KanoWill, 2020-07-30:
Sometimes can't exact, and sometimes get stuck

devttys0, 2020-10-12:
It looks like binwalk is getting "stuck" while attempting to extract the RAR archive. In reality it's probably the external `unrar` utility that is getting stuck, but without the original file that produced this issue it's impossible to say for sure. 

If you run binwalk in debug mode, you'll get a lot more information about what might be happening here:

```$ python -O $(which binwalk) -e test.png```