ReFirmLabs/binwalk #463: Error occurred when extracting an img

cq674350529, 2020-05-22:
```shell
$ binwalk -e rootfs.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             TRX firmware header, little endian, image size: 33026048 bytes, CRC32: 0x86E0F476, flags: 0x0, version: 1, header size: 28 bytes, loader offset: 0x1C, linux kernel offset: 0x1C0000, rootfs offset: 0x0

Signature Exception: not all arguments converted during string formatting
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 596, in main
    retval = self.run()
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 192, in run
    self.scan_file(fp)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 171, in scan_file
    self.result(r=r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 490, in result
    getattr(self, dependency.attribute).callback(r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 222, in callback
    r.name)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 639, in extract
    binwalk.core.common.debug("Extraction successful: %s" % extract_ok)
TypeError: not all arguments converted during string formatting
----------------------------------------------------------------------------------------------------
```

The example image is available at: https://mega.nz/file/3V43mJLJ#rGY8yudESY_AEKC_xKZAMghC1LAHWzbVLEVYfxAe5QQ

Thanks!

cq674350529, 2020-05-22:
It seemed that my binwalk env was broken after upgrade. 
Solved it and closed.

closed 2020-05-22 by cq674350529

EAGLESAT, 2020-05-22:
My brother did not understand me, decipher the file that I sent you and you
will know what I am talking about as it gives me empty files Try it,
disassemble it and you will know

EAGLESAT, 2020-05-22:
This
http://www.mediafire.com/file/mwkuuja0k9bzfvl/GN-RS8-MiniHD-Plus_V2.61_10042020.zip/file

في الجمعة، 22 مايو، 2020 17:52 EAGLE SAT <eaglesat2020@gmail.com> كتب: