ReFirmLabs/binwalk #467: Regression: binwalk result is different between versions v2.2.0 and v2.1.1 (newer seems wrong)

guilhermepiccoli, 2020-06-12:
Hi folks, first let me thank you for binwalk, great and very useful tool.
I've noticed an odd behavior with newer binwalk, when trying to inspect a vmlinux image (Linux kernel image) from powerpc64.

The file I'm trying to inspect is available at: people.canonical.com/~gpiccoli/vmlinux-4.15.0-99-generic.gz
It's compressed, so one must decompress before analysis ("gunzip vmlinux-4.15.0-99-generic.gz"). The md5sum of the vmlinux decompressed is: 48f75557e6bafadf4d2767b725de7ab3

With version 2.1.1, I got the following output: pastebin.ubuntu.com/p/Qj7RmzkPHk
With 2.2.0, the ouput is like: pastebin.ubuntu.com/p/FwXmsBtMSq

I consider the newer version wrong - it shows a bunch of compressed objects, which I tried to extract (the bigger ones at least) using "dd" and failed to decompress them all. It doesn't make sense that output for me...I've bisected using git, and was able to find 2 potential offender commits:

cea0bfbda3 ("Relaxed date validation for gzip signatures")
95447d6751 ("Updated Unix path signatures to prevent common false positives")

Thanks for the attention!

Neustradamus, 2020-08-04:
@devttys0: Have you looked this problem?

devttys0, 2020-10-13:
With the latest code from master I get nearly the exact same output as the v2.2.0 output posted above (which I agree does appear to be the more correct output).

closed 2020-10-13 by devttys0

Neustradamus, 2020-10-13:
@devttys0: The problem is from 2.1.1 vs 2.2.0.

It is normal, badly, that master is like 2.2.0 but it is not like 2.1.1.

Can you do new tests and compare with 2.1.1 and master ?