ReFirmLabs/binwalk #472: Suggest a different Jefferson Repo

ach5948, 2020-06-29:
The jefferson project appears to be abandoned (no commits since 2016). The linked jefferson project only supports python2.7 which reached EOL on 2020-01-01 and is not included in Ubuntu 20.04. Installation of this tool is now quite tedious. At minimum the install instructions for Jefferson should be updated to not rely on the packages python-pip and python-lzma

sviehb, 2020-08-25:
Thanks for bringing this up. jefferson is ported to Python 3.
To make it work with the binwalk deps.sh script MR https://github.com/ReFirmLabs/binwalk/pull/481 needs to be merged.

Ertavf, 2023-02-10:
To maintainer: please update [INSTALL.md#L62](https://github.com/ReFirmLabs/binwalk/blame/master/INSTALL.md#L62)

My thanks to [QKaiser](https://github.com/QKaiser) for letting me know Jefferson has a new maintainer [ONEKEY](https://github.com/onekey-sec/jefferson) and that I can install it via PyPI.

dependencies:

```bash

sudo apt update

sudo apt install liblzo2-dev python3-pip

sudo apt-get install liblzo2-dev

sudo -H pip3 install python-lzo

```

then simply install with

```bash

python3 -m pip install --user jefferson

```

info:

https://github.com/sviehb/jefferson/issues/38#issuecomment-1426440534

closed 2023-09-28 by ach5948