ReFirmLabs/binwalk #484: ubifs image extract failed

cq674350529, 2020-09-08:
When parsing an ubifs image, `binwalk` uses `ubireader_extract_files` to extract. However it may fail sometimes, and the extracted directory contained nothing but empty directory.

```shell
$ binwalk -e 0.ubi

$ tree .
.
├── 0.ubi
└── _0.ubi.extracted
    ├── 0.ubi
    └── ubifs-root
        └── 122063392
            └── kernel    # empty
```

If using `ubireader_extract_images` manually, it will extract files successfully. Then run `binwalk` against the extracted file.
``` shell
$ ubireader_extract_images ./0.ubi

$ tree .
.
├── 0.ubi
└── ubifs-root
    └── 0.ubi
        ├── img-122063392_vol-kernel.ubifs
        └── img-122063392_vol-ubi_rootfs.ubifs
```

krsh, 2021-11-17:
@cq674350529: Is there any news on this topic?

cq674350529, 2021-11-18:
@krsh Currently, I didn't find a proper to integrate both `ubireader_extract_files` and `ubireader_extract_images`. As mentioned above, if `binwalk` failed to extract it, I'll run `binwalk` again after running `ubireader_extract_images`.

krsh, 2021-11-18:
@cq674350529 this solution is not feasible for me, I'm using Binwalk API in an automated tool.

krsh, 2021-11-18:
Pull request: #575

cq674350529, 2021-11-19:
@krsh I saw your pull request, and I'm wondering if it's applicable to all cases. I cames across some firmwares before, which can be extracted with `ubireader_extract_files`, but not with `ubireader_extract_images`.

krsh, 2021-11-19:
@cq674350529 I have the same doubt. Can you provide me with a firmware that works fine with `ubireader_extract_files`?

cq674350529, 2021-11-19:
Currently, I didn't remember which one works fine with `ubireader_extract_files`.  I'll send you if I can find later.

krsh, 2021-11-19:
- this file only works with pull request or `ubireader_extract_images`: https://www.downloads.netgear.com/files/GDC/RAX43/RAX43_V1.0.2.82_2.0.50.zip
- this file only works without pull request or `ubireader_extract_files`:  https://downloads.linksys.com/support/assets/firmware/FW_MR9600_2.0.6.206937_prod.img

krsh, 2021-11-22:
Some notes:

- **UBI + UBIFS**: works with `binwalk` or `ubireader_extract_files`
- **UBI + SquashFS**: doesn't work with `binwalk` and works well with `ubireader_extract_images`

Can someone confirm this analysis?

mzpqnxow, 2024-02-28:
> Some notes:
> 
> 
> 
> - **UBI + UBIFS**: works with `binwalk` or `ubireader_extract_files`
> 
> - **UBI + SquashFS**: doesn't work with `binwalk` and works well with `ubireader_extract_images`
> 
> 
> 
> Can someone confirm this analysis?

It's not much work to enable bibwalk to handle both, though. One way would be to enhance https://github.com/ReFirmLabs/binwalk/blob/master/src/binwalk/plugins/ubivalid.py to utilize ubireader_extract_images and/or ubireader_extract files, rather than arbitrarily choosing one or the other (I noticed one suggested patch just switched from files -> image, I assume that "fixes" some cases at the cost of breaking others)

devttys0, 2024-09-23:
Both UBI and UBIFS images are now supported in Binwalkv3