ReFirmLabs/binwalk #485: Logical shift operators not implemented despite documentation

XXLuigiMario, 2020-09-08:
The Signature File Format wiki page claims that logical shift operators are implemented, but that doesn't appear to be the case.
Attempting to use them results in an unknown data type error, looking at `magic.py` it looks like only "&", "|", "*", "+", "-", "/", "~", "^" are supported.

[image]

llamasoft, 2021-02-27:
I just ran into this issue myself.  The list of supported operators is listed here: https://github.com/ReFirmLabs/binwalk/blob/3154b0012e7dbaf2b20edd5c0a2350ec64009869/src/binwalk/core/magic.py#L100-L119
Noticeably absent are `>>`, `<<`, and `**`.

There's also a bit of misleading wording in the [Signature File Format](https://github.com/ReFirmLabs/binwalk/wiki/Signature-File-Format) wiki that states "Various arithmetic expressions can be applied to both the offset or data type fields."  To me this implies that multiple arithmetic operations can be applied to the data type field (i.e. `ulelong>>4&0x1F`) but this is not the case.

eacmen, 2021-03-16:
fixed in #518

closed 2021-03-16 by eacmen