ReFirmLabs/binwalk #496: binwalk result isn't complete

ymgh96, 2020-11-05:
I know that my [file](https://github.com/ymgh96/CTF-Writeups/blob/main/Boston%20Key%20Party%20CTF%202015/Heath%20Street/secretArchive.6303dd5dbddb15ca9c4307d0291f77f4) has multiple embedded files inside like a KGB file (bellow image confirm it and I see these files on my friend system)
[image]
binwalk output on my friend system:
[image]
But binwalk extract just one main file on my system!
[image]
What's the problem? How do I solve it?

devttys0, 2021-01-24:
The size of the EXT filesystem is reported to be 12582912, which is exactly the size of the secretArchive.6303dd5dbddb15ca9c4307d0291f77f4 file. My guess is the CTF creators did this on purpose, since by default binwalk tries to be smart and skip scanning the contents of the EXT filesystem and in this case it skips all the way to the end of the file.

The solution is to tell binwalk to be dumb: `binwalk --dumb secretArchive.6303dd5dbddb15ca9c4307d0291f77f4`

closed 2021-01-24 by devttys0

ymgh96, 2021-01-24:
Thank you so much. This solution solve my problem!

> The size of the EXT filesystem is reported to be 12582912, which is exactly the size of the secretArchive.6303dd5dbddb15ca9c4307d0291f77f4 file. My guess is the CTF creators did this on purpose, [...]