ReFirmLabs/binwalk #499: sasquatch is now mksquashfs

Secure4Fun, 2020-12-08:
sasquatch is now installed on systems as mksquashfs and unsquashfs. This results in the below errors when running binwalk -e on a suqashfs file.

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -le -d 'squashfs-root' '%e'': [Errno 2] No such file or directory: 'sasquatch', 'sasquatch -p 1 -le -d 'squashfs-root' '%e'' might not be installed correctly

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -be -d 'squashfs-root' '%e'': [Errno 2] No such file or directory: 'sasquatch', 'sasquatch -p 1 -be -d 'squashfs-root' '%e'' might not be installed correctly
1180160       0x120200        Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 2419187 bytes, 544 inodes, blocksize: 131072 bytes, created: 2015-01-05 09:06:52

Errors were also plentiful when trying to run the deps.sh file to build from source. 
Quick work-around for others: sudo ln -s /usr/bin/mksquashfs /usr/bin/sasquatch

p4cx, 2020-12-14:
If it still fails and don't work like expected, here another workaround:
 
Check out [https://github.com/devttys0/sasquatch](https://github.com/devttys0/sasquatch), and then run `CFLAGS=-fcommon ./build.sh`

After that, you will have a functional sasquatch installation on your system.

flamecopper, 2022-02-22:
So how do I erase it if I run into issues

closed 2024-09-23 by devttys0