ReFirmLabs/binwalk #506: Support cmd

RiotGamesU, 2021-01-14:
windows10, thanks

Ayowel, 2021-01-17:
Windows is a WIP, see PR #454 . It should work well enough if you checkout this branch

RiotGamesU, 2021-01-29:
> Windows is a WIP, see PR #454 . It should work well enough if you checkout this branch

Sorry, I don't know how to use GitHub, which branch should I download, thank you

nuno-andre, 2021-01-29:
As long as there is no support for entrypoints in the setup script, you can run binwalk as a module. Just install the package:

```
git clone https://github.com/ReFirmLabs/binwalk.git
cd binwalk
python setup.py install
```

And then, you can run it with `python -m binwalk`. E.g.
```
python -m binwalk -B file.bin
```

If you prefer just a command, after installation, run:
```
cmd -c where binwalk
echo @py.exe ^"%~dp0binwalk^" %* > binwalk.cmd
exit
```

And then you'll be able to run the above example as:
```
binwalk -B file.bin
```

RiotGamesU, 2021-01-30:
> As long as there is no support for entrypoints in the setup script, you can run binwalk as a module. Just install the package:
> 
> ```
> git clone https://github.com/ReFirmLabs/binwalk.git
> cd binwalk
> python setup.py install
> ```
> 
> And then, you can run it with `python -m binwalk`. E.g.
> 
> ```
> python -m binwalk -B file.bin
> ```
> 
> If you prefer just a command, after installation, run:
> 
> ```
> cmd -c where binwalk
> echo @py.exe ^"%~dp0binwalk^" %* > binwalk.cmd
> exit
> ```
> 
> And then you'll be able to run the above example as:
> 
> ```
> binwalk -B file.bin
> ```
> 
> To install the mentioned development branch:
> 
> ```
> git clone --single-branch --branch packaging-fixes https://github.com/akx/binwalk.git
> cd binwalk
> python setup.py install
> ```
> 
> I haven't tried it, but just with this you should have the command available.

thanks, but:
when I run this command:
`C:\program\binwalk\binwalk-2.2.0>python -m binwalk -B C:\Programs\lg.exe
C:\program\python\Python39\python.exe: No module named binwalk.__main__; 'binwalk' is a package and cannot be directly executed`
and:
`C:\program\binwalk\binwalk-2.2.0>cmd -c where binwalk
Microsoft Windows [Version 10.0.19041.746]
(c) 2020 Microsoft Corporation. All rights reserved.

C:\program\binwalk\binwalk-2.2.0>echo @py.exe ^"%~dp0binwalk^" %* > binwalk.cmd

C:\program\binwalk\binwalk-2.2.0>exit

C:\program\binwalk\binwalk-2.2.0>binwalk -B API.md
C:\program1\P\python\Python39\python.exe: can't open file 'C:\program\binwalk\binwalk-2.2.0\binwalk': [Errno 2] No such file or directory`

seems some wrong was happend

nuno-andre, 2021-01-30:
The command error is a conflict with a previous version. You've got another `binwalk` in `%PATH%`, and `where binwalk` finds that path instead of `C:\program\python\Python39\Scripts`.

The first error is also likely to be a version conflict, previous versions were not a package indeed. Maybe you previously installed another version using `pip`? If so, you should uninstall the older version first. In the folder you cloned the repository to:

```
python -m pip uninstall binwalk
python setup.py install
```

And please also delete or rename `C:\program\binwalk` before reinstalling. Then I would try the develop branch first, which will install a `binwalk.exe` in your `Scripts` directory, instead of the current *nix script, and doesn't require further steps.

If you follow the first option, after installation there should be a `binwalk` file in `C:\program\python\Python39\Scripts` with following content:

```
#!/usr/bin/env python
from binwalk.__main__ import main
main()
```

Then, if you can run `py.exe binwalk -B C:\Programs\lg.exe` in the same directory you could proceed with further steps.

RiotGamesU, 2021-01-31:
> The command error is a conflict with a previous version. You've got another `binwalk` in `%PATH%`, and `where binwalk` finds that path instead of `C:\program\python\Python39\Scripts`.
>
> The first [...]

thanks, but my %PATH% does not seem to have binwalk:
`
C:\program\binwalk\binwalk-2.2.0>echo %PATH%
C:\program\VMware\VMware\bin\;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\program\xftp\xftp\;C:\program\xshell\xshell\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\program\java\openjdk\jdk-15.0.1\bin;C:\program\java\JDK11\bin;C:\program\java\JDK15\bin;C:\Program Files\Common Files\Oracle\Java\javapath;%JAVA_HOME14%\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\program\java\JDK8\bin;C:\program\java\JDK8\jre\bin;C:\program\Node.js\Node.js\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files (x86)\IncrediBuild;C:\program\ffmpeg\ffmpeg\bin;C:\program\openssl\openssl\bin;C:\program\m3u8download;C:\program\apktool;C:\program\dex2jar\dex-tools-2.1-SNAPSHOT;C:\program\dex2jar\dex-tools-2.1-SNAPSHOT\lib;C:\program\jadx\jadx-1.1.0\bin;C:\program\nmap\nmap;C:\program\scrcpy\scrcpy-win64-v1.17_2;C:\Program Files\dotnet\;C:\MinGW\bin;C:\MinGW\msys\1.0\bin;C:\MinGW\msys\1.0\sbin;C:\program\Git\Git\cmd;C:\Program Files\PowerShell\7\;C:\Program Files\MySQL\MySQL Server 8.0\bin;C:\program\python\Python39\Scripts\;C:\program\python\Python39\;C:\program\python\python3.8\Scripts\;C:\program\python\python3.8\;C:\Users\xxxxxx\AppData\Local\Microsoft\WindowsApps;C:\program\Bandizip\Bandizip\;C:\program\python\PyCharm\PyCharm 2020.2.2\bin;C:\Users\xxxxxx\AppData\Roaming\npm;C:\program\VSCode\VSCode\bin;C:\program\Kotlin\IntelliJ IDEA Community Edition 2020.2\bin;C:\Users\xxxxxx\.dotnet\tools
`

and when I run:
`
C:\Users\xxxxxx>where binwalk
C:\program\python\Python39\Scripts\binwalk
`

I use this path instead of C:\program\python\Python39\Scripts, but have same error.
and when I uninstall binwalk:
`
C:\program\binwalk\binwalk-2.2.0>python -m pip uninstall binwalk
Found existing installation: binwalk 2.2.0
ERROR: Cannot uninstall 'binwalk'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
`

The content of C:\program\python\Python39\Scripts\binwalk is:
`#!C:\program\python\Python39\python.exe

import os
import sys

# If installed to a custom prefix directory, binwalk may not be in
# the default module search path(s). Try to resolve the prefix module
# path and make it the first entry in sys.path.
# Ensure that 'src/binwalk' becomes '.' instead of an empty string
_parent_dir = os.path.dirname(os.path.dirname(os.path.abspath(__file__)))
for _module_path in [
    # from repo: src/scripts/ -> src/
    _parent_dir,
    # from build dir: build/scripts-3.4/ -> build/lib/
    os.path.join(_parent_dir, "lib"),
    # installed in non-default path: bin/ -> lib/python3.4/site-packages/
    os.path.join(_parent_dir,
                 "lib",
                 "python%d.%d" % (sys.version_info[0], sys.version_info[1]),
                 "site-packages")
]:
    if os.path.exists(_module_path) and _module_path not in sys.path:
        sys.path = [_module_path] + sys.path

import binwalk
import binwalk.modules

def main():
    with binwalk.Modules() as modules:
        try:
            if len(sys.argv) == 1:
                sys.stderr.write(modules.help())
                sys.exit(1)
            # If no explicit module was enabled in the command line arguments,
            # run again with the default signature scan explicitly enabled.
            elif not modules.execute():
                # Make sure the Signature module is loaded before attempting 
                # an implicit signature scan; else, the error message received
                # by the end user is not very helpful.
                if hasattr(binwalk.modules, "Signature"):
                    modules.execute(*sys.argv[1:], signature=True)
                else:
                    sys.stderr.write("Error: Signature scans not supported; ")
                    sys.stderr.write("make sure you have python-lzma installed and try again.\n")
                    sys.exit(2)
        except binwalk.ModuleException as e:
            sys.exit(3)

if __name__ == '__main__':
    try:
        # Special options for profiling the code. For debug use only.
        if '--profile' in sys.argv:
            import cProfile
            sys.argv.pop(sys.argv.index('--profile'))
            cProfile.run('main()')
        else:
            main()
    except IOError:
        pass
    except KeyboardInterrupt:
        sys.stdout.write("\n")
`
Thank you very very much, I will try other branch first.