ReFirmLabs/binwalk #510: STAY AWAY FROM BINWALK PRO

EvoPulseGaming, 2021-01-23:
If it wasn't for the fact they don't charge you up front, I would call this theft.

A: Website is basically empty. Doesn't tell you anything you didn't already know.
B: 3GB file limit
C: 5 uploads per month regardless of size

D: Did I mention is doesn't work? Uploaded two files that regular binwalk works with, and all I get is 0 files extracted.
E: No log, no information about found signatures. If it can't extract a file, you don't get anything.
F: Waiting time is obscene, also no timer or indication of where it is in the process, see G and F.
G: Extracting status can mean unzipping a file, or binwalking it. No information given here.
F: I still have a file that says "extracting" over 24hrs later.

What does work:
Entropy graph........ which is just a picture. for 10$ a month, would be nice to have some grid lines or something like 4k resolution...

If this changes I'm happy to edit this issue. (It will likely be removed anyways)

eacmen, 2021-01-24:
This is not an issue for binwalk community. Any issues for binwalk pro please direct to support@refirmlabs.com. Thank you for your feedback.

closed 2021-01-24 by eacmen

EvoPulseGaming, 2021-01-24:
This is a problem for the community, because of the big banner at the bottom of the read me. How else do you think I ended up at binwalk pro. If the reference to binwalk pro was removed, then it wouldn't be an issue.

eacmen, 2021-01-24:
> This is a problem for the community, because of the big banner at the bottom of the read me. How else do you think I ended up at binwalk pro. If the reference to binwalk pro was removed, then it wouldn't be an issue.
> 
> I really don't appreciate an employee of a company that has a conflicting interest deciding on what is important for a free version of their own product.
> 
> This is so bad, that expecting people to write support to tell them that their product doesn't work is normal.

I don't understand your argument. We created binwalk. We maintain binwalk on behalf of the community. We have volunteered our time for many years making binwalk what it is today. You are free to fork it and maintain your own version if you like. Informing users of a commercial offering of a free community product is a normal and well-established practice.

Either way, using the github issue tracker as a forum for your disapproval of our commercial product is not appropriate. Please take your complaints somewhere else or direct them to support@refirmlabs.com and we will do our best to address them. 

I am locking this issue. I would ask that you please refrain from using the github issue tracker as your own personal soapbox.