ReFirmLabs/binwalk #513: Files identified but not extracted

mrx23dot, 2021-02-10:
Running simple extract
`python binwalk.py --dumb -e test.bin`

founds everything perfectly, but -e cannot extract the files, I have everything in PATH (unzip, jar).
It only extracts one invalid file: 18EADE.zip len: 25722 (none of the listed ones)

When I do it by hand it works perfectly, but very labor some.
Why isn't there a debug-log level what's happening, where it fails to process these block.
**Or a simple option to dump all these findings to binary file.**

[test.zip](https://github.com/ReFirmLabs/binwalk/files/5959133/test.zip)

1632990       0x18EADE        Zip archive data, at least v2.0 to extract, compressed size: 18542, uncompressed size: 173427, name: 90000083-I96CAN_e.xml
1651800       0x193458        Zip archive data, at least v2.0 to extract, compressed size: 810, uncompressed size: 3345, name: 90000083-I96CAN_enum_e.json
1652680       0x1937C8        Zip archive data, at least v2.0 to extract, compressed size: 408, uncompressed size: 2144, name: can_tabl.json
1653132       0x19398C        Zip archive data, at least v2.0 to extract, compressed size: 1441, uncompressed size: 10141, name: descriptor1.json
1654638       0x193F6E        Zip archive data, at least v2.0 to extract, compressed size: 2007, uncompressed size: 28448, name: 90000083-I96CA.json
1656719       0x19478F        Zip archive data, at least v2.0 to extract, compressed size: 513, uncompressed size: 6924, name: 90000083-I96CAN_DIA.json
1657701       0x194B65        End of Zip archive, footer length: 22

Tested on Win64, but that shouldn't matter.

eacmen, 2021-03-16:
The embedded zip file looks malformed.

```
appuser@9d056db20abc:/tmp/_test.zip-0.extracted/_test.bin.extracted$ unzip 18EADE.zip
Archive:  18EADE.zip
warning [18EADE.zip]:  282 extra bytes at beginning or within zipfile
  (attempting to process anyway)
error [18EADE.zip]:  start of central directory not found;
  zipfile corrupt.
  (please check that you have transferred or created the zipfile in the
  appropriate BINARY mode and that you have compiled UnZip properly)
appuser@9d056db20abc:/tmp/_test.zip-0.extracted/_test.bin.extracted$
```

closed 2021-03-16 by eacmen

mrx23dot, 2021-03-16:
Thank you. How did you enable higher log level? I didn't find it in man.

commonism, 2023-12-28:
https://github.com/ReFirmLabs/binwalk/blob/cddfede795971045d99422bd7a9676c8803ec5ee/src/binwalk/core/common.py#L17-L22

```
python3 -O `which binwalk` -Mre …
```