ReFirmLabs/binwalk #523: Update PyPI Package

litneet64, 2021-03-11:
Hi, I'm currently working on a project written in Python that relies on the binwalk API (version `2.2.0`), however that version is not currently available on PyPI which means I cannot upload my package there (newest official one is `2.1.0` released on 2015), as PyPI requires that all required dependencies for a package are already available/accessible on PyPI (aka no direct links to other sites or VCS, as I used before I tried to make a package). 

I know I could make my own fork and then upload my own PyPI package with the newer `2.2.0` version, however this is not a solution in the long run. Release `2.2.0` has been out for a while, so when will it be updated on PyPI too?

diogopim, 2021-09-24:
Hello!

Any developments on this issue? Is there a reason PyPI version is not updated since 2015?

ttufts, 2024-05-13:
Is there any update on this issue? The package on pypi seems to be completely broken now, meaning any dependency on binwalk can't point to pypi. 

Why aren't any new versions being pushed to pypi?

closed 2024-09-23 by devttys0