ReFirmLabs/binwalk #537: deps.sh failing on Kali 2021.1

gregsurber, 2021-04-21:
Trying to install on fresh Kali virtual image, but it fails with E: Unable to locate package qt5base-dev.
I tried removing that from the install script, hoping it would still build without it, maybe just without the pretty GUI. The install then failed on several more dependencies, including:
cramfsprogs
libqt4-opengl
python3-pyqt4
python3-pyqt4.qtopengl
and maybe a few others.

All appear to say either "no installation candidate" or "unable to locate package"

decidedlygray, 2021-04-21:
I submitted a PR for this here - #534 

Something else may be going wrong though because, for example `cramfsprogs` and `libqt4-opengl` should not be installed when the Kali distro is detected (see [here](https://github.com/ReFirmLabs/binwalk/blob/master/deps.sh#L46))

decidedlygray, 2021-04-21:
Also, I recently went through the pain of installing on Kali. Another thing that you're going to find that is broken is the sasquatch dependency, because it doesn't support GCC 10 defaults.

Here is what I did as a workaround:

1. In deps.sh comment out this line: https://github.com/ReFirmLabs/binwalk/blob/master/deps.sh#L87

2. Then manually build and install sasquatch, using the `CFLAGS` workaround for the GCC 10 issue [[source for workaround - open issue in sasquatch project]](https://github.com/devttys0/sasquatch/issues/36):
```
git clone --quiet --depth 1 --branch "master" https://github.com/devttys0/sasquatch
cd sasquatch
CFLAGS=-fcommon ./build.sh
sudo rm -rf sasquatch
```

zi0Black, 2021-04-28:
`(cd sasquatch && ./build.sh)` -> `(cd sasquatch && CFLAGS=-fcommon ./build.sh)` works for me :)

decidedlygray, 2021-04-28:
@zi0Black thanks for confirming that! I tried that first, so I could just have a kind of clean build script that worked (I was deploying multiple machines) but for whatever reason the CFLAGS environment variable didn’t seem to be getting picked up when the tested the script. It ended up failing with the same error. I must have had something wrong though if it works. Admittedly, I was pretty deep down a StackOverflow hole and uninstalling/reinstalling stuff at that point.

flamecopper, 2022-02-22:
After following all the steps above I am still getting:
WARNING: Extractor.execute failed to run external extractor 'unsquashfs -d 'squashfs-root' '%e'': 'module' object has no attribute 'DEVNULL', 'unsquashfs -d 'squashfs-root' '%e'' might not be installed correctly

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -le -d 'squashfs-root' '%e'': 'module' object has no attribute 'DEVNULL', 'sasquatch -p 1 -le -d 'squashfs-root' '%e'' might not be installed correctly

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -be -d 'squashfs-root' '%e'': 'module' object has no attribute 'DEVNULL', 'sasquatch -p 1 -be -d 'squashfs-root' '%e'' might not be installed correctly

WARNING: Extractor.execute failed to run external extractor 'mkdir 'squashfs-root' && mount -t squashfs '%e' 'squashfs-root'': 'module' object has no attribute 'DEVNULL', 'mkdir 'squashfs-root' && mount -t squashfs '%e' 'squashfs-root'' might not be installed correctly
0             0x0             Squashfs filesystem, big endian, lzma signature, version 3.1, size: 4433988 bytes, 1247 inodes, blocksize: 65536 bytes, created: 2011-06-23 10:46:19

closed 2024-09-23 by devttys0