ReFirmLabs/binwalk #545: python: can't open file 'C:\Python27\Scripts\binwalk': [Errno 2] No such file or directory

retr0cube, 2021-05-20:
Hi !

I installed Binwalk from this repo but every time I get this error :

`python: can't open file 'C:\Python27\Scripts\binwalk': [Errno 2] No such file or directory`

Any help is appreciated !

kushkaizer, 2021-10-11:
Hi has anyone been able to solve this problem yet?

ek-nath, 2021-12-13:
I think, generally speaking, you should upgrade the python interpreter from 2.7 to 3.x. There's a note in the main readme as well.

Shengpy, 2022-10-15:
you should open binwalk.py and change the directory where (......\\binwalk-master\\src\\binwalk)  binwalk-master located, then  you open ...\binwalk-master\src\binwalk\modules\extractor.py  and comment all the line have pwd ( #import pwd ), finally binwalk.py will run easily

closed 2024-09-23 by devttys0