ReFirmLabs/binwalk #547: Can't use binwalk

vunhatchuong123, 2021-05-29:
When I try to use binwalk or python binwalk on cmd. It tells me to choose a program to open

megapro17, 2021-06-02:
This shit isn't working in windows

v-micada, 2021-07-09:
**According to the documentation, Binwalk is only experimental on Windows:**
https://github.com/ReFirmLabs/binwalk/wiki/Frequently-Asked-Questions#does-binwalk-work-on-windows
https://github.com/ReFirmLabs/binwalk/wiki/Supported-Platforms
 
**You can try running Binwalk from Cygwin on Windows instead:**
https://github.com/ReFirmLabs/binwalk/wiki/Supported-Platforms
https://www.cygwin.com/

Another option would be to use a Binwalk docker image on Windows.

stoney84, 2021-09-12:
try cmd

Bamgm14, 2021-09-27:
Doesn't work
Revert to Binwalk 2.3.2 and it should be fine
Adding pwd was what caused this

closed 2021-11-02 by vunhatchuong123