ReFirmLabs/binwalk #560: test_firmware_zip fails

swt2c, 2021-09-12:
With binwalk 2.3.3, I'm seeing the zip test fail:

```
======================================================================
FAIL: Test: Open firmware.zip, scan for signatures
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/builddir/build/BUILD/binwalk-2.3.3/testing/tests/test_firmware_zip.py", line 29, in test_firmware_zip
    eq_(len(scan_result[0].results), len(expected_results))
AssertionError: 4 != 2
/builddir/build/BUILDROOT/binwalk-2.3.3-1.fc36.x86_64/usr/lib/python3.10/site-packages/binwalk/modules/extractor.py:969: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if child_pid is 0:
/builddir/build/BUILDROOT/binwalk-2.3.3-1.fc36.x86_64/usr/lib/python3.10/site-packages/binwalk/modules/extractor.py:984: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if child_pid is 0:
```

devttys0, 2021-09-21:
Yes, I noticed this too, it's fixed in commit dd4f2efd275c9dd1001130e82e0f985110cd2754

closed 2021-09-21 by devttys0

swt2c, 2021-09-22:
Actually, that commit seems to be what *caused* the test to fail, rather than fixing it.  That commit appears to be in v2.3.3.  Can you please take another look?

Zsub, 2021-09-28:
v2.3.3 just installed via Homebrew shows this message as well:

```
/usr/local/Cellar/binwalk/2.3.3/libexec/lib/python3.9/site-packages/binwalk/modules/extractor.py:969: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if child_pid is 0:
/usr/local/Cellar/binwalk/2.3.3/libexec/lib/python3.9/site-packages/binwalk/modules/extractor.py:984: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if child_pid is 0:
```