ReFirmLabs/binwalk #578: abort while processing a specific iso file

MMikePL, 2021-11-24:
Binwalk can't cope with public minix iso file, available at http://minix1.woodhull.com/pub/contrib/m203CD08iso.gz
output is interrupted and rest of the input is not processed.

Steps to reproduce error (example based on Windows 10 machine):
1. download minix iso from http://minix1.woodhull.com/pub/contrib/m203CD08iso.gz
2. unzip downloaded file to c:\minix2.0.3CD0.8.iso (ie with 7zip)
3. create new fat32 50MB partition as M:
4. copy exe file
copy %windir%\explorer.exe m:\
5. copy iso file
copy c:\minix2.0.3CD0.8.iso m:\
6. copy another two exe files
copy %windir%\notepad.exe m:\
copy %windir%\regedit.exe m:\
7. dump whole partition to file (i've used HxD tool)
8. analyze file by binwalk (default settings)
binwalk part_dump.bin | grep "executable"

Above steps produce result with only one line, not three.

The same steps but omitting step 5 (don't copy iso file) produce 3 lines in result (correct).

Dump file is too big to attach directly (even after compression), i've uploaded it to https://www.sendspace.com/file/t20q21

I understand that there may be sth wrong with this iso, but binwalk should (at least) ignore it and parse rest of the input...

mzpqnxow, 2024-02-03:
You can have it carve the file out rather than try to extract it, or you can instruct binwalk to keep invalid files, as a few ideas. Read all of the [command line flags](https://github.com/ReFirmLabs/binwalk/wiki/Usage) it’s not a long read

mzpqnxow, 2024-02-03:
I’ll add that `7z x minix.iso` should work really well for you, too. In general I don’t think binwalk is the best tool for well known files like rar, zip, iso, etc. They already have native tools that work just fine (though i would still be curious to know what the underlying issue is)

devttys0, 2024-10-03:
Hmmm, not sure what was causing this, but Binwalkv3 seems to handle the Minix ISO fine.

closed 2024-10-03 by devttys0