ReFirmLabs/binwalk #592: WebP support?

mhombach, 2022-01-26:
I was wondering if binwalk already supports finding `webp` files in a dump/bin and if not, if that would be something easy to be implemented :)

mzpqnxow, 2024-02-03:
Type “webp” into the github search box for this project, that will probably answer your first question 

I can’t speak to the second, but generally speaking if webp files have a “magic” and/or well-known header, then no, it’s not terribly difficult. Though you may have noticed this repository is no longer maintained so you’re on your own (see Binwalk pro)

devttys0, 2024-09-23:
Binwalk v3 now supports WebP image detection and extraction:

[image]

closed 2024-09-23 by devttys0