ReFirmLabs/binwalk #594: windows 10 ModuleNotFoundError: No module named 'pwd'

babyshen, 2022-02-24:
```
# windows 10
# python setup.py install 
C:\software\Python37\Scripts>python binwalk -h
Traceback (most recent call last):
  File "binwalk", line 4, in <module>
    __import__('pkg_resources').run_script('binwalk==2.3.3', 'binwalk')
  File "C:\software\Python37\lib\site-packages\pkg_resources\__init__.py", line 664, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "C:\software\Python37\lib\site-packages\pkg_resources\__init__.py", line 1444, in run_script
    exec(code, namespace, namespace)
  File "c:\software\python37\lib\site-packages\binwalk-2.3.3-py3.7.egg\EGG-INFO\scripts\binwalk", line 2, in <module>
    from binwalk.__main__ import main
  File "C:\software\Python37\lib\site-packages\binwalk-2.3.3-py3.7.egg\binwalk\__main__.py", line 24, in <module>
    import binwalk.modules
  File "C:\software\Python37\lib\site-packages\binwalk-2.3.3-py3.7.egg\binwalk\modules\__init__.py", line 16, in <module>
    from binwalk.modules.extractor import Extractor
  File "C:\software\Python37\lib\site-packages\binwalk-2.3.3-py3.7.egg\binwalk\modules\extractor.py", line 7, in <module>
    import pwd
ModuleNotFoundError: No module named 'pwd'

```

why ??

babyshen, 2022-02-24:
binwalk 2.3.3

nmantani, 2022-02-24:
Though I created the pull request https://github.com/ReFirmLabs/binwalk/pull/577 to fix this error in November 2021, it has not been merged yet. 😥

babyshen, 2022-02-25:
binwalk 2.3.2 works fine on windows10.

DieselDX, 2022-02-28:
Is it work now? How do you solve the problem? Same error occur to me.

babyshen, 2022-02-28:
> Is it work now? How do you solve the problem? Same error occur to me.

use binwalk 2.3.2

babyshen, 2022-02-28:
@DieselDX    use binwalk 2.3.2

kazegh, 2022-05-01:
I have the same question:How do you solve the problem?
Windows 11
Python 3.9.10 (tags/v3.9. , Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)] on win32

hahn23, 2022-05-17:
I used the following workaround. Download the 2.3.2 package and take the file "binwalk-2.3.2.zip\binwalk-2.3.2\src\binwalk\modules\extractor.py" and replace it with the one under "C:\software\Python37\lib\site-packages\binwalk-2.3.3-py3.7.egg\binwalk\modules\" (according your example)

After that it should work with 2.3.3 as well.

CypherpunkSamurai, 2022-05-17:
For someone looking to run binwalk without all the editing and stuff, I've made a fork of biwalk with PR #577 merged, and with unofficial released binaries.

https://github.com/CypherpunkSamurai/binwalk-win/releases/tag/v2.3.4

nmantani, 2022-05-18:
@CypherpunkSamurai Thank you for using my fix!

No commit or merge have been made to this repository since last September...  Is binwalk abandoned?

CypherpunkSamurai, 2022-05-18:
> @CypherpunkSamurai Thank you for using my fix!

Thanks to you for the PR, I was going through the PR for a fix then i found this issue :)

> No commit or merge have been made to this repository since last September... Is binwalk abandoned?

Binwalk is probably the primary tool in most people's re toolkits. I doubt it's abandoned. The devs are probably busy with other projects / life.

Byxs20, 2022-08-23:
> 我使用了以下解决方法。下载2.3.2包，将“binwalk-2.3.2.zip\binwalk-2.3.2\src\binwalk\modules\extractor.py”文件替换为“C:\software\Python37\ lib\site-packages\binwalk-2.3.3-py3.7.egg\binwalk\modules”（根据你的例子）
> 
> 它之后也应该与2.3.3一起使用。

Thank you. That solved my problem.

dwisuseno, 2023-06-12:
> > 我使用了以下解决方法。下载2.3.2包，将“binwalk-2.3.2.zip\binwalk-2.3.2\src\binwalk\modules\extractor.py”文件替换为“C:\software\Python37\ lib\site-packages\binwalk-2.3.3-py3.7.egg\binwalk\modules”（根据你的例子）
> > 它之后也应该与2.3.3一起使用。
> 
> Thank you. That solved my problem.

I don't get it, how to solve this issue about no module pwd.

closed 2024-09-23 by devttys0