ReFirmLabs/binwalk #596: ubi filesystem extraction

MatthewTingum, 2022-03-23:
From what I can tell, binwalk doesn't support extraction of `ubi` filesystems.

I'm not certain if there's a magic that can be used to detect them.
There are python tools that can be used to extract ubi files such as [ubidump](https://github.com/nlitsme/ubidump)

What would it take to include such a tool?

mzpqnxow, 2024-02-03:
I will push the changes I made to my fork when I have time, or you can make them yourself

Add an `extractor` method to the class @  https://github.com/ReFirmLabs/binwalk/blob/master/src/binwalk/plugins/ubivalid.py 

Have it use the ubifs tools. Follow the [cpio plugin](https://github.com/ReFirmLabs/binwalk/blob/master/src/binwalk/plugins/ubivalid.py) as a guide

If I remember correctly the command I used for the extraction in that method was `ubi_image_extract`

devttys0, 2024-09-23:
UBI is supported in Binwalk v3:

[image]

closed 2024-09-23 by devttys0