ReFirmLabs/binwalk #603: 5.16.0-12parrot1-amd64 pkg_resources.DistributionNotFound Error

CryptoKevinM, 2022-06-27:
When trying to run binwalk (2.3.1) on 5.16.0-12parrot1, I receive the following stack trace for a pkg_resources.DistributionNotFound error:

```
/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 1.14.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 581, in _build_master
    ws.require(__requires__)
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 909, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 800, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (binwalk 2.2.1 (/usr/lib/python3/dist-packages), Requirement.parse('binwalk==2.3.3+fa0c0bd'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/binwalk", line 4, in <module>
    __import__('pkg_resources').run_script('binwalk==2.3.3+fa0c0bd', 'binwalk')
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3260, in <module>
    def _initialize_master_working_set():
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3234, in _call_aside
    f(*args, **kwargs)
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3272, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 596, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/home/user/.local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 795, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'binwalk==2.3.3+fa0c0bd' distribution was not found and is required by the application
```

CryptoKevinM, 2022-06-27:
Here is the install output for when I run `sudo python3 setup.py install`:

```
creating 'dist/binwalk-2.3.3+fa0c0bd-py3.9.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing binwalk-2.3.3+fa0c0bd-py3.9.egg
removing '/usr/lib/python3.9/site-packages/binwalk-2.3.3+fa0c0bd-py3.9.egg' (and everything under it)
creating /usr/lib/python3.9/site-packages/binwalk-2.3.3+fa0c0bd-py3.9.egg
Extracting binwalk-2.3.3+fa0c0bd-py3.9.egg to /usr/lib/python3.9/site-packages
/usr/lib/python3.9/site-packages/binwalk-2.3.3+fa0c0bd-py3.9.egg/binwalk/modules/extractor.py:969: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if child_pid is 0:
/usr/lib/python3.9/site-packages/binwalk-2.3.3+fa0c0bd-py3.9.egg/binwalk/modules/extractor.py:984: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if child_pid is 0:
binwalk 2.3.3+fa0c0bd is already the active version in easy-install.pth
Installing binwalk script to /usr/bin

Installed /usr/lib/python3.9/site-packages/binwalk-2.3.3+fa0c0bd-py3.9.egg
Processing dependencies for binwalk==2.3.3+fa0c0bd
Finished processing dependencies for binwalk==2.3.3+fa0c0bd
```

NicolasFNino, 2024-06-06:
are there any updates on this issue?

closed 2024-09-23 by devttys0