ReFirmLabs/binwalk #604: Failed to load plugin module 'gzipextract'

mkczyk, 2022-07-05:
## Error when using -D

I try to extract files from example file [TL-WR841N_V8_130506.zip](http://www.tp-link.com/resources/software/TL-WR841N_V8_130506.zip) (from [tutorial](https://github.com/briankip/binwalk-tutorial)):
```
binwalk -D='*' 'wr841nv8_en_3_13_33_up_boot(130506).bin'
```
and I get error:
```
WARNING: Failed to load plugin module 'gzipextract': 'NoneType' object has no attribute 'search'
```

## Detailed steps to reproduce

Two options (on Ubuntu and on Docker for 100% reproducibility), but I get the same result.

### On Ubuntu

1. Install binwalk
```
wget https://github.com/ReFirmLabs/binwalk/archive/refs/tags/v2.3.3.zip
unzip binwalk-2.3.3.zip
cd binwalk-2.3.3
sudo python3 setup.py install
```
notice that I get errors on installation:
```
byte-compiling build/bdist.linux-x86_64/egg/binwalk/modules/extractor.py to extractor.cpython-38.pyc
build/bdist.linux-x86_64/egg/binwalk/modules/extractor.py:969: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if child_pid is 0:
build/bdist.linux-x86_64/egg/binwalk/modules/extractor.py:984: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if child_pid is 0:
```

2. Download example file:

```
wget http://www.tp-link.com/resources/software/TL-WR841N_V8_130506.zip
unzip TL-WR841N_V8_130506.zip
```

3. It works:

```
$ binwalk 'wr841nv8_en_3_13_33_up_boot(130506).bin'

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             TP-Link firmware header, firmware version: 0.-16396.3, image version: "", product ID: 0x0, product version: 138477576, kernel load address: 0x0, kernel entry point: 0x80002000, kernel offset: 4063744, kernel length: 512, rootfs offset: 813084, rootfs length: 1048576, bootloader offset: 2883584, bootloader length: 0
13392         0x3450          U-Boot version string, "U-Boot 1.1.4 (May  6 2013 - 13:20:35)"
13440         0x3480          CRC32 polynomial table, big endian
14728         0x3988          uImage header, header size: 64 bytes, header CRC: 0xBA7F2047, created: 2013-05-06 05:20:35, image size: 34860 bytes, Data Address: 0x80010000, Entry Point: 0x80010000, data CRC: 0x263C3839, OS: Linux, CPU: MIPS, image type: Firmware Image, compression type: lzma, image name: "u-boot image"
14792         0x39C8          LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 99104 bytes
131584        0x20200         TP-Link firmware header, firmware version: 0.0.3, image version: "", product ID: 0x0, product version: 138477576, kernel load address: 0x0, kernel entry point: 0x80002000, kernel offset: 3932160, kernel length: 512, rootfs offset: 813084, rootfs length: 1048576, bootloader offset: 2883584, bootloader length: 0
132096        0x20400         LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 2317284 bytes
1180160       0x120200        Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 2652846 bytes, 537 inodes, blocksize: 131072 bytes, created: 2013-05-06 05:32:12
```

4. But it doesn't works:

```
$ binwalk -D='*' 'wr841nv8_en_3_13_33_up_boot(130506).bin'

WARNING: Failed to load plugin module 'gzipextract': 'NoneType' object has no attribute 'search'

WARNING: Failed to load plugin module 'gzipextract': 'NoneType' object has no attribute 'search'

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             TP-Link firmware header, firmware version: 0.-16396.3, image version: "", product ID: 0x0, product version: 138477576, kernel load address: 0x0, kernel entry point: 0x80002000, kernel offset: 4063744, kernel length: 512, rootfs offset: 813084, rootfs length: 1048576, bootloader offset: 2883584, bootloader length: 0
13392         0x3450          U-Boot version string, "U-Boot 1.1.4 (May  6 2013 - 13:20:35)"
13440         0x3480          CRC32 polynomial table, big endian
14728         0x3988          uImage header, header size: 64 bytes, header CRC: 0xBA7F2047, created: 2013-05-06 05:20:35, image size: 34860 bytes, Data Address: 0x80010000, Entry Point: 0x80010000, data CRC: 0x263C3839, OS: Linux, CPU: MIPS, image type: Firmware Image, compression type: lzma, image name: "u-boot image"
14792         0x39C8          LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 99104 bytes
131584        0x20200         TP-Link firmware header, firmware version: 0.0.3, image version: "", product ID: 0x0, product version: 138477576, kernel load address: 0x0, kernel entry point: 0x80002000, kernel offset: 3932160, kernel length: 512, rootfs offset: 813084, rootfs length: 1048576, bootloader offset: 2883584, bootloader length: 0
132096        0x20400         LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 2317284 bytes
1180160       0x120200        Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 2652846 bytes, 537 inodes, blocksize: 131072 bytes, created: 2013-05-06 05:32:12
```

### On Docker

1. Install binwalk

```
wget https://github.com/ReFirmLabs/binwalk/archive/refs/tags/v2.3.3.zip
unzip binwalk-2.3.3.zip
cd binwalk-2.3.3
docker build -t binwalk .
```

2. Download example file:

```
wget http://www.tp-link.com/resources/software/TL-WR841N_V8_130506.zip
unzip TL-WR841N_V8_130506.zip
```

3. It works:
```
docker run -v "$(pwd):/home/appuser" binwalk 'wr841nv8_en_3_13_33_up_boot(130506).bin'
```

4. But it doesn't works:
```
docker run -v "$(pwd):/home/appuser" binwalk -D='*' 'wr841nv8_en_3_13_33_up_boot(130506).bin'
```

securitybites, 2023-04-05:
I also have this same issue

inanity04, 2023-10-15:
also having this issue on MacOS Ventura 13.5.2

PeteHaughie, 2024-05-28:
Same issue here

closed 2024-09-23 by devttys0