ReFirmLabs/binwalk #610: pkg_resources.VersionConflict: (binwalk 2.1.0 (/home/lenovo/.local/lib/python3.8/site-packages), Requirement.parse('binwalk==2.3.3+fa0c0bd'))

samannazz, 2022-08-19:
```
lenovo@lenovo-Lenovo-ideapad-330-15IKB:~$ binwalk
/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 0.1.36ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 0.23ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
Traceback (most recent call last):
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 581, in _build_master
    ws.require(__requires__)
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 909, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 800, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (binwalk 2.1.0 (/home/lenovo/.local/lib/python3.8/site-packages), Requirement.parse('binwalk==2.3.3+fa0c0bd'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/binwalk", line 4, in <module>
    __import__('pkg_resources').run_script('binwalk==2.3.3+fa0c0bd', 'binwalk')
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3260, in <module>
    def _initialize_master_working_set():
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3234, in _call_aside
    f(*args, **kwargs)
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3272, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 596, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/home/lenovo/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 795, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'binwalk==2.3.3+fa0c0bd' distribution was not found and is required by the application

```
what is this Error about? I have installed binwalk and uninstall it many times but the error is the same.

a4865g, 2022-08-22:
Perhaps you have previously used pip to install to binwalk.
Just uninstall it, and reinstall.

```
$ python -m pip uninstall binwalk
```

```
$ sudo python3 setup.py uninstall
$ sudo ./deps.sh
$ sudo python3 setup.py install
```

samannazz, 2022-08-23:
I have uninstalled binwalk:
`python -m pip uninstall binwalk`

and there is no setup.py file,
and if i run command binwalk it gives same error

a4865g, 2022-08-23:
The setup.py file is on this github.

You just need to git clone and go into the repo folder and do what I did earlier, it should work.

closed 2024-09-23 by devttys0