ReFirmLabs/binwalk #618: binwalk firmware extract issue in Ubuntu 22.04 LTS

iAmG-r00t, 2022-11-02:
So apt install of binwalk has an issue of missing `sasquatch` install. I went ahead and purged it, installed binwalk from this repo followed the guide faced this [issue](https://github.com/devttys0/sasquatch/issues/48) fixed it as suggested [here](https://github.com/devttys0/sasquatch/issues/48#issuecomment-1267506233) but now when I try to extract this [binary](http://legacyfiles.us.dlink.com/DIR-655/REVC/FIRMWARE/DIR-655_REVC_FIRMWARE_3.02.B05.ZIP) the files in 'extract_dir/squashfs-root/www/' are empty.

But when I install binwalk in Ubuntu 20.04.5, first I don't face any errors, two the files are extracted properly and are not empty. Ohhh and there aren't two `squashfs-root` directories.

Can someone try and see what the issue can be?

## My Issue

[image]

[image]

## Expectations

[image]

[image]

iAmG-r00t, 2022-11-02:
I hope I have explained it properly.

jacopotediosi, 2023-02-16:
To correctly compile sasquatch on Ubuntu 22 apply https://github.com/devttys0/sasquatch/pull/51 instead of https://github.com/devttys0/sasquatch/pull/47:

```
cd /tmp
git clone --quiet --depth 1 --branch "master" https://github.com/devttys0/sasquatch
cd sasquatch
wget https://github.com/devttys0/sasquatch/pull/51.patch && patch -p1 <51.patch
sudo ./build.sh
```

Doing so, I see files extracted from your firmware aren't empty anymore.

nrathaus, 2023-10-16:
Thank you for the help @jacopotediosi - this should be made part of the INSTALL process for binwalk.. not sure why its not

Cossack9989, 2024-07-11:
lol https://github.com/devttys0/sasquatch/pull/51.patch, the patch also works on Debian12