ReFirmLabs/binwalk #623: Different names for `cramfsck`

rel-s, 2023-01-10:
The default configuration is as stated:
```
# Try cramfsck first; if that fails, swap the file system and try again
^cramfs filesystem:cramfs:cramfsck -x '%%cramfs-root%%' '%e':0:False
^cramfs filesystem:cramfs:cramfsswap '%e' '%e.swap' && cramfsck -x '%%cramfs-root%%' '%e.swap':0:False
```
However, on some distributions (tested on debian WSL2) the command has been renamed to `fsck.cramfs`, and the `-x` argument to `--extract`. Perhaps we should try both if one fails.

Shorrer, 2023-08-26:
Another problem is that `fsck.cramfs` is in `/usr/sbin` on Debian (tested on Debian 12.1), which is not in the `PATH` of non-root users.

h00die-gr3y, 2024-05-10:
@Shorrer ,
Easiest way to get around the problem is to download the `cramfsck` from https://github.com/npitre/cramfs-tools and compile for your distribution. Tested on Kali Linux AARCH64 and it works perfectly fine.

closed 2024-09-23 by devttys0