ReFirmLabs/binwalk #650: Symlink Error

lilsheepyy, 2023-07-07:
Hello, well my problem is that im trying to extract the netgear r6100 v1.0.1.28 firmware, when I run this command:
sudo binwalk -e R6100-V1.0.1.28.img --run-as=root
To extract everything I got a lot of symlink error that says they point outside the extraction folder here is the prompt:
sheep@sheep:~/Downloads/r6100$ sudo binwalk -e R6100-V1.0.1.28.img --run-as=root

sheep@sheep:~/Downloads/r6100$ sudo binwalk -e R6100-V1.0.1.28.img --run-as=root

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
192           0xC0            LZMA compressed data, properties: 0x6D, dictionary size: 8388608 bytes, uncompressed size: 3346944 bytes

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/var -> /tmp; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/home -> /tmp; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/www -> /tmp/www; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/wpa2 -> /tmp/etc/wpa2; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/mtab -> /proc/7032/mounts; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/hosts -> /tmp/hosts; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/passwd -> /tmp/config/passwd; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/group -> /tmp/config/group; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/ppp -> /tmp/ppp; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/gshadow -> /tmp/config/gshadow; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/fstab -> /tmp/fstab; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/resolv.conf -> /tmp/resolv.conf; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/samba -> /tmp/samba; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/ath -> /tmp/etc/ath; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/passwd+ -> /tmp/config/passwd+; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/TZ -> /tmp/TZ; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/shadow -> /tmp/config/shadow; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/etc/config/wireless -> /tmp/etc/wireless; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/dni_apup -> /tmp/sbin/dni_apup; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/ipconflict -> /usr/sbin/net-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/daemonv6 -> /usr/sbin/net-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/coexistence_2.4G -> /usr/sbin/coexist-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/cmdsched_wlan_status -> /usr/sbin/net-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/cfg -> /usr/sbin/cgiMain; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/button_test_mode -> /usr/sbin/button-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/coexistence_5G -> /usr/sbin/coexist-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/rs_send -> /usr/sbin/net-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/detwanv6 -> /usr/sbin/net-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/cmdsched -> /usr/sbin/net-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/show_coexistence -> /usr/sbin/coexist-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/show_all_button_status -> /usr/sbin/button-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/sbin/cmdroute -> /usr/sbin/net-util; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/usr/private -> /tmp/private; changing link target to /dev/null for security purposes.

WARNING: Symlink points outside of the extraction directory: /home/sheep/Downloads/r6100/_R6100-V1.0.1.28.img.extracted/squashfs-root/usr/lib/smb.conf -> /tmp/samba/smb.conf; changing link target to /dev/null for security purposes.

[... cut ...]
1179776       0x120080        Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 10025227 bytes, 1526 inodes, blocksize: 131072 bytes, created: 2018-06-29 15:02:00

As you can see a lot of files point "outside" the extraction folder, which I guess it is some sort of an error and that they are actually trying to point inside but binwalk recognizes it as it is not and launches some security measures, Im trying to learn about all this and can´t find anyone with this same error.

zmxuee, 2024-06-21:
Me too,I got a information :add "--preserve-symlinks" option to ignore these warngings;
I'm wondering why these warnings appear :<

zmxuee, 2024-06-21:
> Hello, well my problem is that im trying to extract the netgear r6100 v1.0.1.28 firmware, when I run this command: sudo binwalk -e R6100-V1.0.1.28.img --run-as=root To extract everything I got a lot [...]

Me too,I got a information :add "--preserve-symlinks" option to ignore these warngings;
I'm wondering why these warnings appear :<

closed 2024-09-23 by devttys0