ReFirmLabs/binwalk #651: ubireader problem

urn1ce, 2023-07-21:
https://github.com/jrspruitt/ubi_reader doesn't have setup.py, but in https://github.com/ReFirmLabs/binwalk/blob/master/deps.sh still use setup.py
function install_ubireader
{
    git clone --quiet --depth 1 --branch "master" https://github.com/jrspruitt/ubi_reader
    (cd ubi_reader && $SUDO $PYTHON setup.py install)
    $SUDO rm -rf ubi_reader
}

urn1ce, 2023-07-21:
https://github.com/ReFirmLabs/binwalk/blob/master/INSTALL.md also have this problem

nhtri2003gmail, 2023-07-24:
Hi, I see this link seems useful: https://github.com/ReFirmLabs/binwalk/issues/642#issuecomment-1567514863

urn1ce, 2023-07-24:
> Hi, I see this link seems useful: [#642 (comment)](https://github.com/ReFirmLabs/binwalk/issues/642#issuecomment-1567514863)

This is very useful, but I still hope that the official team will fix it.

dierye, 2023-08-15:
That's because ubi_reader use a new branch name "main".If you don't want to change a lot , you can easily change the branch param to "v0.8.5-master".

kdelson, 2023-09-21:
Agree, this should be a pretty simple fix but it appears the ubi_reader team have migrated to poetry so anything relying on the old setup script is broken in latest.

ojuanjo, 2024-03-21:
When I changed the command in the sh file, he had a new problem:

Processing ./ubi_reader
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
ERROR: Could not find a version that satisfies the requirement lzallright<0.3.0,>=0.2.1 (from ubi-reader) (from versions: none)
ERROR: No matching distribution found for lzallright<0.3.0,>=0.2.1

closed 2024-09-23 by devttys0