ReFirmLabs/binwalk #657: Would it be possible to use the built-in python module 'getpass' as a somewhat OS agnostic way to get the username?

JeffreyO, 2023-08-30:
As opposed to 'pwd' which is Unix specific and/or needing to litter the Python code with conditional statements about the OS.

JeffreyO, 2023-08-30:
As for the test for Admin or root, I haven't found an OS agnostic way yet. The closest I could find was this answer from [grigoryvp](https://stackoverflow.com/users/69882/grigoryvp) on this question on Stack Overflow: https://stackoverflow.com/questions/1026431/cross-platform-way-to-check-admin-rights-in-a-python-script-under-windows

I'm still looking.

mzpqnxow, 2024-02-03:
Well, the purpose of checking if binwalk is running as root/administrator, why not theow away the idea that this should be done based on a username and instead use pseido-ish code like:

    def rootdir():
        if windows: return “c:\”
        return “/“
    if can_write(os.path.join(rootdir(), “testfile”)):
        raise RuntimeError(“too dangerous”)
        
Checking for the username “Administrator” or “root” is not really a sufficient check to determine safety anyway

I’m not much of a Windows user, but I can speak to Linux and say that there are ways to allow any user to bypass filesystem access controls so (e.g. CAP_* capabilities that override kernel filesystem permission checking logic)

Having logic like to actually test if the dangerous behavior is permitted with a real test can simplify the check and make it more comprehensive on all of Windows, Linux and MacOS.

closed 2024-09-23 by devttys0