ReFirmLabs/binwalk #659: binwalk fails to extract after filename/extension confusion

whoami-defnotme, 2023-09-22:
While trying to run `binwalk -eM` against an image, `binwalk` fails with the following error:

```
Signature Exception: [Errno 2] No such file or directory: '/tmp/_IMAGE.extracted/_OFFSET.extracted/FILENAME'
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/binwalk-2.3.3-py3.11.egg/binwalk/core/module.py", line 595, in main
    retval = self.run()
             ^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/binwalk-2.3.3-py3.11.egg/binwalk/modules/signature.py", line 190, in run
    for fp in iter(self.next_file, None):
  File "/usr/local/lib/python3.11/site-packages/binwalk-2.3.3-py3.11.egg/binwalk/core/module.py", line 419, in next_file
    fp = self.config.open_file(next_target_file)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/binwalk-2.3.3-py3.11.egg/binwalk/modules/general.py", line 197, in open_file
    return binwalk.core.common.BlockFile(fname,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/binwalk-2.3.3-py3.11.egg/binwalk/core/common.py", line 523, in BlockFile
    return InternalBlockFile(fname, mode=mode, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/binwalk-2.3.3-py3.11.egg/binwalk/core/common.py", line 358, in __init__
    super(self.__class__, self).__init__(fname, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/_IMAGE.extracted/_OFFSET.extracted/FILENAME'
```

If I list the contents of the directory containing `FILENAME`, it contains:

```
> ls /tmp/_IMAGE.extracted/_OFFSET.extracted/
...
FILENAME.gz
```

So somehow the file is being written with the `gz` extension but is then referenced using only the filename without the `gz` extension.

This reproduces reliably locally on macOS as well as inside a Docker container using an image built via the Dockerfile in this repo (after fixing #658 locally on my machine). I've sanitised file names/paths for now as I'm unsure if the image is shareable.

hackeyz, 2023-09-22:
code works fine as long as you run it as:

`$ binwalk --run-as=root -e filename`

whoami-defnotme, 2023-09-23:
Unfortunately, it does not.

```
# First, we use the command provided:
> binwalk --run-as=root -e IMAGE

Extractor Exception: In order to execute third party applications as root, binwalk must be run with root privileges.
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/binwalk/2.3.4/libexec/lib/python3.11/site-packages/binwalk/core/module.py", line 258, in __init__
    self.load()
  File "/opt/homebrew/Cellar/binwalk/2.3.4/libexec/lib/python3.11/site-packages/binwalk/modules/extractor.py", line 160, in load
    raise ModuleException("In order to execute third party applications as %s, binwalk must be run with root privileges." % self.runas_user)
binwalk.core.exceptions.ModuleException: In order to execute third party applications as root, binwalk must be run with root privileges.
----------------------------------------------------------------------------------------------------

# So now we elevate our privileges and get the error on the exact same file which triggered this initial report
# Note: many other things extract successfully, as expected
> sudo binwalk -eM --run-as=root IMAGE
...
Signature Exception: [Errno 2] No such file or directory: '/tmp/_IMAGE.extracted/_OFFSET.extracted/FILENAME'
...traceback...
```

hackeyz, 2023-09-26:
did you run command in the exact folder containing the file?

closed 2024-09-23 by devttys0