ReFirmLabs/binwalk #667: Python 3.12 compatibility issue: No module named 'imp'

mzpqnxow, 2024-01-24:
Python 3 deprecated `imp` and as of 3.12 (possibly earlier) it's removed

```
8:22:48 › ~/.local/bin/binwalk -e 240
Traceback (most recent call last):
  File "/home/binwalk/.local/bin/binwalk", line 2, in <module>
    from binwalk.__main__ import main
  File "/home/binwalk/.local/lib/python3.12/site-packages/binwalk/__init__.py", line 3, in <module>
    from binwalk.core.module import Modules
  File "/home/binwalk/.local/lib/python3.12/site-packages/binwalk/core/module.py", line 18, in <module>
    import binwalk.core.plugin
  File "/home/binwalk/.local/lib/python3.12/site-packages/binwalk/core/plugin.py", line 4, in <module>
    import imp
ModuleNotFoundError: No module named 'imp'
```

It's easy to fix, for those encountering this I'll send a PR

Python >= 3.11 or 3.12 have some significant performance enhancements so it's worthwhile addressing this in my opinion

FroggMaster, 2024-03-17:
Currently encountering this issue myself, would love for this to be merged. :)

ElDavoo, 2024-03-21:
Please merge, thank you

mzpqnxow, 2024-03-30:
> Hi! :wave: 
> 
> 
> 
> I'm currently investigating packages still using the `imp` module, which has been [removed with Python 3.12](https://docs.python.org/dev/whatsnew/3.12.html#imp), as we are about to upgrade to Python 3.12 on Arch Linux.
> 
> 
> 
> It would be great if https://github.com/ReFirmLabs/binwalk/pull/668 could be merged and released, as it would help us in upgrading! :)

FYI, you should probably consider switching packages off of this repo and to the one that is referenced in this issue/PR

As I understand it, this repo is no longer maintained, the community fork has effectively taken its place

You could of course stick with this repo for whatever reason, but with the number of unmerged PRs- many being bugfixes- you may find yourself managing a list of patches from the fork anyway 😀

closed 2024-09-23 by devttys0