ReFirmLabs/binwalk #671: Dead Project - Parent Company Killed by Microsoft :( - See readme

redawl, 2024-02-28:
Anyone know of any alternatives / a maintained fork? Gentoo maintainers are looking to drop this in March, since the project appears to be dead, and the dependencies are becoming more and more out of date :/

stkw0, 2024-02-28:
Hi. I made a fork (here https://github.com/OSPG/binwalk) to try to gather the community around and maintain the open source version (that seems to be sadly abandoned). I currently maintain a couple of packages on Gentoo and my initial intend was precisely to keep binwalk alive in Gentoo while at the same time have feedback from different maintainers and experienced users.

redawl, 2024-02-29:
> Hi. I made a fork (here https://github.com/OSPG/binwalk) to try to gather the community around and maintain the open source version (that seems to be sadly abandoned). I currently maintain a couple [...]

Thanks for picking up the project for Gentoo! I don't have any open source experience, but if you need/want any help with maintenance, let me know :)

stkw0, 2024-03-01:
I think it would be better to send the PR to the fork so it's easy to keep track of which patches have already been applied and also it's easier to avoid merge conflicts. Also, I am trying to get in touch with different packagers of different Linux distributions to work together so much more users would benefit from it.

AndrewFasano, 2024-03-10:
[Unblob](https://github.com/onekey-sec/unblob) seems to be a decent alternative. I did some testing and found it to beat binwalk in terms of performance and extraction success rates. The project is actively maintained and they (quickly) fixed a number of bugs I reported.

stkw0, 2024-03-10:
Seems an interesting alternative, although seems it supports less data types than binwalk. Maybe some people could try to port some missing features to unblob so we don't need binwalk anymore. Until then I will keep maintaining binwalk

devttys0, 2024-09-23:
https://github.com/ReFirmLabs/binwalk/tree/binwalkv3

[image]

closed 2024-09-23 by devttys0